me-no-dev/ESPAsyncWebServer #377: HTTPS or TLS / SSL support ?

JacoFourie, 2018-07-02:
This is more a question.
Does the server support HTTPS ?

arihantdaga, 2018-08-07:
Yes it does.. 
Although examples are not written properly but you can look into code and see functions like beginSecure()

bbasil2012, 2018-08-17:
And what about examples ?

me-no-dev, 2018-08-19:
HTTPS is only for ESP8266 for now, and I do not recommend using it, because it limits the server to one concurrent connection and causes heap to drop dramatically...

me21, 2018-11-20:
Are there any plans for supporting HTTPS on ESP32?

kamil2234, 2018-11-20:
I join to this question

me21, 2018-12-03:
There's an implementation of HTTPS server for ESP32: https://github.com/fhessel/esp32_https_server
Unfortunately, it does not support neither templates nor HTML forms.

closed 2019-10-05 by stale[bot]

s00500, 2019-12-27:
Any update on this ?