me-no-dev/ESPAsyncWebServer #383: What can I use instead of delay?

rjlexx, 2018-07-14:
Hello, @me-no-dev ! 

I have a lot of request hadlers which uses delay. What can I use instead of delay? Or how can I hadle requests by other way with hadlers which uses delay?
It looks like IRsend also uses delay in "send" functions and I also can not use it with ESPAsyncWebServer.

For example:

void handleDisableTv(AsyncWebServerRequest *request) {
  for (int i = 0; i < 3; i++) {
    irsend.sendRC6(TV_DISABLE_CODE, TV_CODE_LEN);
    delay(100);
  }
  request->send(200, "text/plain", F("OK"));
}

void handleIrsend(AsyncWebServerRequest *request) {
  // /irsend/type/code/bits
  String uri = request->url();
  int startPosType = uri.indexOf(F("/irsend/")) + 8;
  int endPosType = uri.indexOf(F("/"), startPosType);
  String type = uri.substring(startPosType, endPosType);
  
  int startPosCode = uri.indexOf(F("/"), endPosType) + 1;
  int endPosCode = uri.indexOf(F("/"), startPosCode);
  String codeStr = uri.substring(startPosCode, endPosCode);
  
  int startPosBits = uri.indexOf(F("/"), endPosCode) + 1;
  int endPosBits = uri.indexOf(F(" "), startPosBits);
  String bitsStr = uri.substring(startPosBits, endPosBits);
  
  int bits = bitsStr.toInt();
  if (type == F("rc5")) {
    int code = codeStr.toInt();
    irsend.sendRC5(code, bits);
  }
  if (type == F("rc6")) {
    int code = codeStr.toInt();
    irsend.sendRC6(code, bits);
  }
  if (type == F("nec")) {
    unsigned long code = strtoul(codeStr.c_str(), NULL, 16); 
    irsend.sendNEC(code, bits);
  }
    
  request->send(200, "text/plain", "OK");
}

hreintke, 2018-07-14:
Hi,

You can use schedule to run functions with delay().
The scheduled functions are executed after each loop().

I have an "empty loop" so for me the execution is almost immediate but differs on application design.
You can't use the  `AsyncWebServerRequest *request` as that is gone but something like this will work.

```
void handleDisableTv(AsyncWebServerRequest *request) {
schedule_function(
[](){
   for (int i = 0; i < 3; i++) {
     irsend.sendRC6(TV_DISABLE_CODE, TV_CODE_LEN);
    delay(100);
  }
});
request->send(200, "text/plain", F("OK"));
}
```

or 
```
schedule_function(std::bind(&irclass::send,irsend,code,bits));
```
change irclass to the class of irsend to make that work.

rjlexx, 2018-07-14:
Hello, @hreintke!

Thank you very much for the fast response! It's exactly what I need. Works perfectly!
Remarks:
With this code 
schedule_function(std::bind(&IRsend::send,irsend,code,bits));
Compiling fails with the error:
could not convert 'std::bind(_Func&&, _BoundArgs&& ...) ........ to 'std::function<void()>'
I've created local function "sendCommandIR" and this code compiles fine:

closed 2019-10-05 by stale[bot]