me-no-dev/ESPAsyncWebServer #387: how to?

dmytro-nff, 2018-07-18:
Hi!
I want to make a call file upload from my library ...
help is needed...
how can I make an analogy (Setup global and class functions as request handlers in your manual):

void handleRequest(AsyncWebServerRequest *request)
{ }
...
void begin(){

		// attach global request handler
		classWebServer.on("/example", HTTP_ANY, handleRequest);

		// attach class request handler
		classWebServer.on("/example", HTTP_ANY, std::bind(&WebClass::classRequest, this, std::placeholders::_1));
	}

my file upload function with params:
void onUpload(AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, boolean final)
{
...
}
Thank you!

dmytro-nff, 2018-07-19:
I understood how it works!
Thank you!

closed 2018-07-19 by dmytro-nff

ogass81, 2018-10-01:
How did you do it?