me-no-dev/ESPAsyncWebServer #388: Best Way to change WiFi Network by http request?

Kameeno, 2018-07-19:
Hi! i'm looking for the best way to change the connected network on esp8266 by an HTTP request, 
for the moment i got this (i tried lot of times and ways but the results is always a big crash).
so: 
```
server.on("/wifi/connect", HTTP_GET, [](AsyncWebServerRequest *request) {
    CoolAsyncEditor coolAsyncEditor;
    int params = request->params();
    for (int i = 0; i < params; i++) {
      AsyncWebParameter *p = request->getParam(i);
      if (!p->isFile() && !p->isPost()) {
        Serial.printf("GET[%s]: WiFi:%s\n", p->name().c_str(),
                      p->value().c_str());
        CoolAsyncEditor coolAsyncEditor;
        if (coolAsyncEditor.getSavedWifi(p->value()) == "") {
          request->send(404);
        } else {
          AsyncWiFiManager wifiManager(&server, &dns);
          if (wifiManager.autoConnect(
                  coolAsyncEditor.getSavedCredentialFromIndex(atoi(p->value().c_str() ),
                                                              "ssid").c_str(),
                  coolAsyncEditor.getSavedCredentialFromIndex(atoi(p->value().c_str() ),
                                                              "pass").c_str() ) ){
          request->send(200, "text/json","CONNECTED!");
          }
          else {
            request->send(200, "text/json", "Not connected :(");
          }
        }
      }
    }
  });
```
coolAsyncEditor is lib inspired by SPIFFSEditor lib to have custom access to SPIFFS, it's working good and she's dont' have problems, i can send files in page, add new wifi credentials inside a json  etc.. 
but... how i can reconnect to another wifi Network properly? Thanks a lot guys.

Kameeno, 2018-07-19:
@Pablo2048  Thanks pablo, i prefere tu don't use AsyncWifiManager in any case, you have any idea about how i can do a WiFi.begin() and verify if the connection is done or not ?

Pablo2048, 2018-07-19:
As I said - write function which connects to the AP in STA mode, call it in schedule, but beware of two things:
1. ESP8266 is single radio, so you probably loose connection to the ESP when switching channels (at least temporarily)
2. you can not return status of connection this way (async callback has to be done as fast as possible - no wait, no delay, etc.)

3rwww1, 2018-07-19:
Suggested scenario:
- use the HTTP request to store some state variable indicating we wish to connect to a new network
- let the main loop poll that variable and trigger a connection to the new AP
- send the connection result to the client through SSE

`delay()` would thus be available, thus allowing enough time for the Wifi connection to be established.

In other words:
- HTTP server doesn't do anything but forwarding "requests" to the main loop by modifying a global state (might be a singleton)
- The main loop react to those change in state by triggering an action (connect to a different Wifi network)
- When the action is completed, the status is sent to the client using SSE (however, this would be complicated by network change due to connection loss, but by no means impossible - SSE client would loose the connection, then have to reconnect, at which point one could send the current Wifi connection state as the first payload of the SSE handshake)

Would that work in your opinion @Pablo2048  ?

Pablo2048, 2018-07-19:
Yep :-) Better explanation than my attempt...

Pablo2048, 2018-07-19:
One slight modification - do not use any variable (and/or flag) - this speeds down your loop() - use schedule_function() one-time trigger.

Kameeno, 2018-07-27:
hey! i arrive to get change my wifi connection, i'm using a singleton class called `CoolAsyncWiFiAction`

inside void loop(){}
i have  `CoolAsyncWiFiAction::getInstance().manageConnectionPortal();` 
where inside i manage where to connect, etc... 
but.. i'm getting a big crash with exeption 29 (why? i got 29kb of free ram)
when i try to send a http response 
what is wrong here? 
```
      AsyncWebServerRequest *request;
      AsyncWebServerResponse *response =
      request->beginResponse(200, "text/plain", "CONNECTED!");
      response->addHeader("CoolBoard", "Cool Async Web Server");
      request->send(response);

```

Kameeno, 2018-07-27:
i can confirm if i use `events.send("connected!", NULL, millis(), 1000);` it's working 
but is not what i'm looking for...

Pablo2048, 2018-07-27:
Hi,
if I got you right, you can not use "request" object this way - pointer to TCP socket part is not filled in. You can not get connection from server side to client in HTTP protocol. Probably best way to do what you need is Websockets - it is persistent connection between server and client and you can write to (established) socket any time from both ends (javascript in web browser, or Arduino in server).

Kameeno, 2018-07-27:
thanks @Pablo2048 !

piotrtutrbo, 2019-05-19:
Hello :)
I have been looking for a few days how to use the great AsyncWifiManager ... for a moment to connect to another IP ... send a few commands there and return to the original AP.
Seemingly banal ... but it gives me a lot of trouble. Several experiments ended with resetting the settings ..

1) I can not use: wifiManager.disconnect; ?
2) How to configure the new IP and password in the scetch? :

Do I understand correctly that connecting to a new IP requires an ESP reset?
(sorry, I'm a beginner but I'm already delighted with the asynchronous approach to the program structure)
Regards

closed 2019-10-05 by stale[bot]