me-no-dev/ESPAsyncWebServer #390: Unable to include AsyncJson in different locations

dskw, 2018-07-30:
Since d7399a7664bce76e9939837fd6cf51bd272ef588 I am unable to have an include to AsyncJson in different files and run into the following linking error:
```
Archiving built core (caching) in: /tmp/arduino_cache_751141/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_e35129e817913bbe969ad3c4750c592e.a
sketch/GameEditor.cpp.o:(.data.JSON_MIMETYPE+0x0): multiple definition of `JSON_MIMETYPE'
sketch/GameBoard.ino.cpp.o:(.data.JSON_MIMETYPE+0x0): first defined here
sketch/GameManager.cpp.o:(.data.JSON_MIMETYPE+0x0): multiple definition of `JSON_MIMETYPE'
sketch/GameBoard.ino.cpp.o:(.data.JSON_MIMETYPE+0x0): first defined here
```

The mentioned cpp heads look like this:
GameManager.cpp:
```
#include "GameManager.h"
#include "Utils.h"
#include <AsyncJson.h>
#include <ArduinoJson.h>
```

GameEditor.cpp:
```
#include "GameEditor.h"
#include "Utils.h"
#include <Hash.h>
#include <AsyncJson.h>
#include <ArduinoJson.h>
```

**Proposed solution:**
In AsyncJson.h change from this:
`const char* JSON_MIMETYPE = "application/json";`
to this:
`constexpr char* JSON_MIMETYPE = "application/json";`

tfelix, 2018-07-31:
I can confirm exactly the same error. I am no C++ expert but it seems the string inclusion in the header is done the wrong way (see: https://stackoverflow.com/questions/2879548/is-it-appropriate-to-set-a-value-to-a-const-char-in-the-header-file) and will break if the AsyncJson.h is included multiple times (which can easy happen if you for example use libraries like Basecamp for the ESP32).

Using your tip to works but gives me a warning: ISO C++ forbids converting a string constant to a char* 
So I vote for `const char* const JSON_MIMETYPE = "application/json";` which works fine.

0xC0DEBA5E, 2018-08-06:
Some people who are trying to use [basecamp](https://github.com/merlinschumacher/Basecamp/issues/63) are also facing issues because the definition of `JSON_MIMETYPE` is causing conflicts.

I tried both solutions from above and both seemed to work.

trycoon, 2018-12-01:
Would it be possible to get this merged soon? My application is split between several files that needs to import the AsyncJson.h and I get
```
Linking .pioenvs/nodemcuv2/firmware.elf
.pioenvs/nodemcuv2/src/main.cpp.o:(.data.JSON_MIMETYPE+0x0): multiple definition of `JSON_MIMETYPE'
.pioenvs/nodemcuv2/src/api.cpp.o:(.data.JSON_MIMETYPE+0x0): first defined here
collect2: error: ld returned 1 exit status
```

PR #396 (2018-12-02): Corrected datatype für JSON_MIMETYPE
This fixes #390
commit 95dedf7a (2018-12-02): Corrected datatype für JSON_MIMETYPE (#396)

* Change JSON MIME type to application/json

* Corrected datatype für JSON_MIMETYPE (Fixes #390)
closed 2018-12-02 by me-no-dev