me-no-dev/ESPAsyncWebServer #392: 'class AsyncServer' has no member named 'beginSecure'

eliasmagn, 2018-08-02:
Cant compile with beginSecure!
```
#include <ESP8266WiFi.h>
 #include <FS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "SD.h"
#include "user_interface.h"
#include <DNSServer.h>
#define ASYNC_TCP_SSL_ENABLED
AsyncWebServer server(80);
AsyncWebServer httpsserver(443);
.
.
.
.
  server.begin();
  httpsserver.beginSecure("/server.cer", "/server.key", NULL);
```
nore error: 'class AsyncWebServer' has no member named 'onSslFileRequest'

```
  request->redirect("/index.htm");
//drawGraph(request);
  });

  httpsserver.onSslFileRequest([](void * arg, const char *filename, uint8_t **buf) -> int {
    Serial.printf("SSL File: %s\n", filename);
    File file = SPIFFS.open(filename, "r");
    if(file){
      size_t size = file.size();
      uint8_t * nbuf = (uint8_t*)malloc(size);
      if(nbuf){
        size = file.read(nbuf, size);
        file.close();
        *buf = nbuf;
        return size;
      }
      file.close();
    }
    *buf = 0;
    return 0;
  }, NULL);

  httpsserver.onNotFound([](AsyncWebServerRequest *request){
```
Any ideas or a hint?
thx in advance!

me21, 2018-12-03:
EspAsyncWebServer doesn't support HTTPS on ESP32 (yet).

closed 2019-10-05 by stale[bot]

jsonpoindexter, 2020-04-18:
@eliasmagn `ASYNC_TCP_SSL_ENABLED` needs to be declared before the imports of ESPAsyncTCP and ESPAsyncWebServer like so:

```
#define ASYNC_TCP_SSL_ENABLED true
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
```

eliasmagn, 2020-04-19:
Nice! I hardcoded this somehow back then. But this was ugly. How easy it could have been wouldnt i have stopped thinking about... Thank you!