me-no-dev/ESPAsyncWebServer #401: 'AcSSlFileHandler' has not been declared

squonk11, 2018-08-20:
I am trying to activate SSL on the ESPAsyncWebServer. For this I set ASYNC_TCP_SSL_ENABLED=1. After that I get 
`/msys32/opt/devel/AsyncWebServer/components/AsyncWebServer/ESPAsyncWebServer.h:398:27: error: 'AcSSlFileHandler' has not been declared
     void onSslFileRequest(AcSSlFileHandler cb, void* arg);`
I can not find the referenced function "AcSSlFileHandler" - neither in the sources of ESPAsyncWebServer nor in the sources of AsyncTCP. Do you have a hint for me?

me21, 2018-12-03:
ESPAsyncWebServer doesn't support HTTPS on ESP32 (yet).

ViniP8, 2019-09-18:
Any news about the HTTPS support on ESP32?

me21, 2019-11-17:
Please keep it open until it's resolved or decided that it won't be fixed.

closed 2020-01-30 by stale[bot]