me-no-dev/ESPAsyncWebServer #402: AsyncCallbackJsonWebHandler does not trigger

poengAlex, 2018-08-24:
The AsyncCallbackJsonWebHandler does not trigger upon url request.

```c
AsyncCallbackJsonWebHandler* handler = new AsyncCallbackJsonWebHandler("/endpoint", [](AsyncWebServerRequest *request, JsonVariant &json) {
	JsonObject& jsonObj = json.as<JsonObject>();
	jsonObj.prettyPrintTo(Serial);
	request->send(200, "application/json", "{test: \"ok\"}");
});
server.addHandler(handler);
```
```javascript
$.ajax({ 
        type: 'GET', 
        url: '/endpoint', 
        data: { 
            txt: "test",
            nr: 123,
            digi: true
        }, 
        dataType: 'json',
        contentType: 'application/json; charset=UTF-8'
    })
    .done(function(data) {
        console.log( "success" );
        console.log(data);
        $scope.checkConnection();
    })
    .fail(function(data) {
        console.log( "error" );
        console.log(data);
    })
    .always(function() {
        console.log( "complete" );
    });
```
```
_HEADER[Host]: 192.168.0.193
_HEADER[Connection]: keep-alive
_HEADER[Pragma]: no-cache
_HEADER[Cache-Control]: no-cache
_HEADER[Accept]: application/json, text/javascript, */*; q=0.01
_HEADER[X-Requested-With]: XMLHttpRequest
_HEADER[User-Agent]: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/67.0.3396.87 Safari/537.36 OPR/54.0.2952.64
_HEADER[Content-Type]: application/json; charset=UTF-8
_HEADER[Referer]: http://192.168.0.193/
_HEADER[Accept-Encoding]: gzip, deflate
_HEADER[Accept-Language]: en-US,en;q=0.9
Host: 192.168.0.193
NOT_FOUND: GET http://192.168.0.193/endpoint
_GET[SSID]: test
_GET[password]: 200
_GET[digi]: true
```

brunohorta82, 2018-09-17:
Hello I have the same issue I think it's because cors :(,

poengAlex, 2018-09-17:
> Hello I have the same issue I think it's because cors :(,

Why? The JS file is served from the same server as the ajax request.

brunohorta82, 2018-09-17:
yes, if I use Postman, works fine, but if I use jQuery no :(

brunohorta82, 2018-09-17:
Finally I solved the problem :)
change this contentType: 'application/json; charset=UTF-8' to contentType: 'application/json'

:)

jnicolson, 2019-01-29:
I just had the same problem, but I resolved it by modifying

https://github.com/me-no-dev/ESPAsyncWebServer/blob/95dedf7a2df5a0d0ab01725baaacb4f982dedcb2/src/AsyncJson.h#L126

changing equalsIgnoreCase to startsWith (both Arduino string functions).  Probably isn't the best solution (be better to parse the Content-Type properly and just get the first bit to compare) but it let me continue

warlord0, 2019-03-28:
> I just had the same problem, but I resolved it by modifying
> 
> [ESPAsyncWebServer/src/AsyncJson.h](https://github.com/me-no-dev/ESPAsyncWebServer/blob/95dedf7a2df5a0d0ab01725baaacb4f982dedcb2/src/AsyncJson.h#L126)
> 
> Line 126 in [95dedf7](/me-no-dev/ESPAsyncWebServer/commit/95dedf7a2df5a0d0ab01725baaacb4f982dedcb2)
> 
>  if (!request->contentType().equalsIgnoreCase(JSON_MIMETYPE)) 
> changing equalsIgnoreCase to startsWith (both Arduino string functions). Probably isn't the best solution (be better to parse the Content-Type properly and just get the first bit to compare) but it let me continue

Good catch. Saved me some work. In my case I was able to modify my client call to pass only the mimetype with no code page. Meaning I didn't have to modify the `AsyncJson.h` file. But I can see this being a problem where I don't control the client script.

closed 2019-10-05 by stale[bot]

famzah, 2023-05-29:
I also was getting a 404 File Not Found error (firing the onNotFound() handler) because of this "content-type" check.

If you asked me, the check is misleading and should be removed. My jQuery AJAX requests were done with (the incorrect) "application/x-www-form-urlencoded; charset=UTF-8" content-type header, but nevertheless they were working fine on my local Linux PHP server. Additionally, regardless of the "content-type" header data, we still can receive an invalid JSON body. I don't see how the "content-type" header check helps in any way.