me-no-dev/ESPAsyncWebServer #412: Requesting several static files from SPIFFS hangs

borisneubert, 2018-09-09:
Hi,

we have ESPAsyncWebServer on ESP8266 serving a web page that requests several more javascript and css files (JQuery etc.) (static from SPIFFS). ESPAsyncWebServer hangs when the third or fourth file is requested. We worked around that issue by merging all javascript files into a single large one. 

The issue is obscured by caching - the second load succeeds because the other files which were previously loaded are already in the browser cache and are not requested again. One can nicely observe the issue with the webdeveloper tools of the browser in the network monitoring tab.

Probably related: #349 

I put some debbuging output into the ESPAsyncWebServer code but it did not make sense to me. How can I solve this issue?

Kind regards,
Boris

MatCat, 2018-10-07:
I am also having a major issue with this, I switched on my project to use this hoping it would solve the issue of trying to load the various images, CSS, JS, etc, but it is just as limited and failing in the same way as the normal webserver code :(

borisneubert, 2018-11-24:
Is there anything I could do to help solving this issue?

closed 2019-10-05 by stale[bot]

Dacesilian, 2020-01-06:
I think this issue still exists and should be resolved somehow :(