me-no-dev/ESPAsyncWebServer #413: class AsyncWebServer' has no member named 'setAuthentication'

michelep, 2018-09-10:
As titled, i got an error while using setAuthentication on "server" :

AsyncWebServer server(80);
[....]
 server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html").setTemplateProcessor(templateProcessor);
  
  // If present, set authentication
  if(strlen(config.www_username) > 0) {
    server.setAuthentication(config.www_username, config.www_password);
  }

error: 'class AsyncWebServer' has no member named 'setAuthentication'
     server.setAuthentication(config.www_username, config.www_password);

GeorgeFlorian, 2019-05-09:
Hello !
Did you manage to fix it ?

closed 2019-10-05 by stale[bot]

Uinden, 2023-02-17:
Solution is here https://github.com/me-no-dev/ESPAsyncWebServer/issues/991#issuecomment-1434751178