me-no-dev/ESPAsyncWebServer #418: Combining with WiFiManager.

HamzaHajeir, 2018-10-05:
Hello

I've tried to use both AsyncTCP and WiFiManager at one sketch, but got many errors.

I guess it comes because of libraries overlapping, I am not sure .

I wonder to know how to use both libraries at one sketch ?

Error : 

```

Arduino: 1.8.7 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:56:18: error: redeclaration of 'HTTP_GET'

   HTTP_GET     = 0b00000001,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:29: note: previous declaration 'HTTPMethod HTTP_GET'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                             ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:57:18: error: redeclaration of 'HTTP_POST'

   HTTP_POST    = 0b00000010,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:39: note: previous declaration 'HTTPMethod HTTP_POST'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                       ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:58:18: error: redeclaration of 'HTTP_DELETE'

   HTTP_DELETE  = 0b00000100,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:72: note: previous declaration 'HTTPMethod HTTP_DELETE'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                                        ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:59:18: error: redeclaration of 'HTTP_PUT'

   HTTP_PUT     = 0b00001000,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:50: note: previous declaration 'HTTPMethod HTTP_PUT'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:60:18: error: redeclaration of 'HTTP_PATCH'

   HTTP_PATCH   = 0b00010000,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:60: note: previous declaration 'HTTPMethod HTTP_PATCH'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                            ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:61:18: error: 'HTTP_HEAD' redeclared as different kind of symbol

   HTTP_HEAD    = 0b00100000,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:9:0:

C:\Users\Hamza\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:25:12: error: previous declaration of 'const char HTTP_HEAD [143]'

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

            ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:62:18: error: redeclaration of 'HTTP_OPTIONS'

   HTTP_OPTIONS = 0b01000000,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:85: note: previous declaration 'HTTPMethod HTTP_OPTIONS'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                                                                                     ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:

C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:63:18: error: redeclaration of 'HTTP_ANY'

   HTTP_ANY     = 0b01111111,

                  ^

In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:

C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:19: note: previous declaration 'HTTPMethod HTTP_ANY'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                   ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Pablo2048, 2018-10-05:
Try to use https://github.com/alanswx/ESPAsyncWiFiManager ...

HamzaHajeir, 2018-10-07:
Thanks @Pablo2048  

It works 👍

lbussy, 2019-09-13:
It's not a matter of better, it's a matter of a derivative art correctly showing it's provenance.  It's "legal" because of the original licensing, but it's rude.  It also prevents one from easily merging in any improvements from the original.  Finally, and perhaps worse, the instructions he's left on the project are verbatim from the original.  If someone were to follow them they would end up getting the original art without any idea why it was not working with the Async libraries.

Anyone is always free to roll their own, however, the strength of the ecosystem is a library of maintained, common code available for reuse in other projects.  Dead-ends lead to death.

I'm not sure I have time to fix his work, my interests lie in the core project with which I am working, not fixing someone else's project.  I guess I have decisions to make in that case.

Pablo2048, 2019-09-13:
Well, you have to write a note to Alan first instead of all of this IMO. Strength of the ecosystem is in maintained libraries, but if everybody has "interests in the core project ... not fixing someone else's project" then there would be no libraries at all.

lbussy, 2019-09-13:
I absolutely do not have to write a note to Alan.  His very first Issue [#1 Curious but not forked](https://github.com/alanswx/ESPAsyncWiFiManager/issues/1) was about this and he's chosen not to do anything about it since July 4, 2016.  Odds are a "note" from me will not change things.

And your last comment is myopic as well.  Clearly, some people's core projects ARE those libraries that they maintain and from which we all benefit.

"All this" is you objecting to me calling his work a mess.  The core question I posed was the existence of a library that worked with this project to present a captive wifi configuration portal.  Apparently, the only two which exist are dead-end projects with no provenance and no courtesy given to the prior art.  I choose not to use those, so I'll head a different way.

consolacion, 2019-10-17:
"Mess" or not, I understand it works, so in lieu of anything 'better'  or less 'messy', it is the best around and I will use it.

lbussy, 2019-12-16:
I am still very interested in adding WiFiManager (or similar) support to this project.

HamzaHajeir, 2019-12-16:
@lbussy  Check [ESPArto Framework](https://github.com/philbowles/esparto) 

You can find your interests there.

lbussy, 2019-12-16:
Thank you @HamzaHajeir, I'll have a look.

I think the original ask is still valid however.  If one has an existing project, having a library change is easier than rebuilding on a new framework.  There might be something there I can easily leverage.

HamzaHajeir, 2019-12-17:
@lbussy That's right. 

I've spent time converting a project to ESPArto.

You can consider it for upcoming projects.

lbussy, 2020-02-27:
Still an issue, despite the bot's best intention.

lbussy, 2020-04-27:
If anyone is curious (or masochistic) [I forked](https://github.com/lbussy/AsyncWiFiManager) WiFiManager and made it work with ESPAsyncWebServer.

If you want to use it, fork it, please.  Right now I'm not intending to take issues anyone else may hit.  "It Works for Me™."

After reflection, this issue would be best raised in WiFiManager, so I'll refrain from arguing with Stale Bit anymore.

alirezaimi, 2020-06-27:
I have this problem too, any solution ?

thisua, 2020-07-24:
> 
> 
> Hello
> 
> I've tried to use both AsyncTCP and WiFiManager at one sketch, but got many errors.
> 
> I guess it comes because of libraries overlapping, I am not sure .
> 
> I wonder to know how to use both libraries at one sketch ?
> 
> Header files included :
> 
> ```
> #include <ESP8266WiFi.h>
> #include <DNSServer.h>
> #include <ESP8266WebServer.h>
> #include <WiFiManager.h>       
> #include <stdio.h>
> #include <malloc.h>
> #include <string>
> #include <Arduino.h>
> #include <Hash.h>
> #include <ESPAsyncTCP.h>
> #include <ESPAsyncWebServer.h>
> #include "ArduinoJson.h"
> ```
> 
> Error :
> 
> ```
> 
> Arduino: 1.8.7 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:
> 
> C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:56:18: error: redeclaration of 'HTTP_GET'
> 
>    HTTP_GET     = 0b00000001,
> 
>                   ^
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:
> 
> C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:29: note: previous declaration 'HTTPMethod HTTP_GET'
> 
>  enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
> 
>                              ^
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:
> 
> C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:57:18: error: redeclaration of 'HTTP_POST'
> 
>    HTTP_POST    = 0b00000010,
> 
>                   ^
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:
> 
> C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:39: note: previous declaration 'HTTPMethod HTTP_POST'
> 
>  enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
> 
>                                        ^
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:16:0:
> 
> C:\Users\Hamza\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:58:18: error: redeclaration of 'HTTP_DELETE'
> 
>    HTTP_DELETE  = 0b00000100,
[... 78 lines cut ...]
> 
>    HTTP_ANY     = 0b01111111,
> 
>                   ^
> 
> In file included from C:\Users\Hamza\Desktop\Programming developments\PlayColor Arduino\AsyncTCP___AsyncJson___WiFiManager\AsyncTCP___AsyncJson___WiFiManager.ino:8:0:
> 
> C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:19: note: previous declaration 'HTTPMethod HTTP_ANY'
> 
>  enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
> 
>                    ^
> 
> exit status 1
> Error compiling for board NodeMCU 1.0 (ESP-12E Module).
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
> ```

> ESP8266WebServer

hi .   Workbench on a single platform name only
one is ESPAsyncWebServer or ESP8266WebServer

egrekov, 2020-08-03:
Hi!
Faced the same problem, but analysis of the code showed that if the WEBSERVER_H directive is declared, then HTTP_GET is ignored ...

```c++
#ifndef WEBSERVER_H
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111,
} WebRequestMethod;
#endif
```

Therefore, to solve the problem, you need to do the following in your code:
```c++
#include "WiFiManager.h"
WiFiManager wifiManager;

#define WEBSERVER_H
#include "WebServerHandler.h"
```

xuanloct4, 2020-08-15:
I tried the  @egrekov solution, it works.
The problem caused by conflicting between ESPAsyncWebServer and ESP8266WebServer libraries.
In order to solve it, we just need to define WEBSERVER_H
  #define WEBSERVER_H
before import ESPAsyncWebServer lib
#include "ESPAsyncWebServer.h"

gibo77, 2020-08-15:
I have used both with success before. You just have to use different ports for each of them. So if ESPAsynchWebserver uses 8080 then WifiManager has to use different such as 80. 
But for my uses case WifiManager did not fit. So I created mymown WifiManager with smashing success. I was happy with it and beast part ESPAsynchWebserever runs in tandem with it. The ESP goes into AP mode while ESPAsynchWebserver not yet activated. Key is make a while loop in setup while user is punching in the passwordnusername.

egrekov, 2020-08-16:
@gibo77 The problem is not in the same ports, the fact is that two libraries define the same types, which leads to a compilation error, but the authors of `ESPAsynchWebserever `provided a solution to this problem.
PS. It would be nice if it was described in the documentation.

lbussy, 2020-08-16:
> the authors of `ESPAsynchWebserever `provided a solution to this problem.

@gibo77 - link please?

egrekov, 2020-10-16:
@lbussy [ESPAsyncWebServer.h#60](https://github.com/me-no-dev/ESPAsyncWebServer/blob/f13685ee97675be2ac9502d177d3024ebc49c1e0/src/ESPAsyncWebServer.h#L60)

HamzaHajeir, 2020-12-16:
I think the time has came to close this issue, As it seems to me.

closed 2020-12-16 by HamzaHajeir

egrekov, 2020-12-17:
Hi! Sorry, but I suggest you reopen the problem.
It was necessary to use this library on esp8266 and process HTTP methods, and then I realized that HTTP constants do not give the values I would like.

```c++
#include <Arduino.h>

#include "WiFiManager.h"
WiFiManager wifiManager;

#define WEBSERVER_H
#include "ESPAsyncWebServer.h"

#define HTTP_CONTENT_TYPE "text/html; charset=utf-8"

AsyncWebServer server(80);

void setup(){
  Serial.begin(115200);

  wifiManager.setConfigPortalTimeout(60);
  wifiManager.autoConnect("wifi-test");

  server.on("/test/", [](AsyncWebServerRequest *request) {
    Serial.printf("REQUEST METHOD: %d\n", request->method());
    Serial.printf("REQUEST METHOD STRING: %s\n", request->methodToString());
    Serial.printf("\nHTTP_CONSTANT:\nHTTP_GET %d\n", HTTP_GET);
    Serial.printf("HTTP_POST %d\n", HTTP_POST);
    Serial.printf("HTTP_DELETE %d\n", HTTP_DELETE);
    Serial.printf("HTTP_PUT %d\n", HTTP_PUT);
    Serial.printf("HTTP_PATCH %d\n", HTTP_PATCH);
    Serial.printf("HTTP_HEAD %d\n", HTTP_HEAD);
    Serial.printf("HTTP_OPTIONS %d\n", HTTP_OPTIONS);
    Serial.printf("HTTP_ANY %d\n", HTTP_ANY);
    request->send(200, HTTP_CONTENT_TYPE, "OK");
  });

  server.begin();
}

void loop(){
}
```
I send a PUT request to esp8266 and receive, request method != HTTP constant:
```
REQUEST METHOD: 8
REQUEST METHOD STRING: PUT

HTTP_CONSTANT:
HTTP_GET 1
HTTP_POST 3
HTTP_DELETE 6
HTTP_PUT 4
HTTP_PATCH 5
HTTP_HEAD 2
HTTP_OPTIONS 7
HTTP_ANY 0
```

I send a PUT request to ESP32 and receive, request method == HTTP constant:
```
REQUEST METHOD: 8
REQUEST METHOD STRING: PUT

HTTP_CONSTANT:
HTTP_GET 1
HTTP_POST 2
HTTP_DELETE 4
HTTP_PUT 8
HTTP_PATCH 16
HTTP_HEAD 32
HTTP_OPTIONS 64
HTTP_ANY 127
```

`#define WEBSERVER_H` should solve the problem of compiling and declaring constants, but it doesn't.

[ESPAsyncWebServer.h#L60](https://github.com/me-no-dev/ESPAsyncWebServer/blob/f6fff3f91ebf45b91ca4cff2460d2febd61f1e27/src/ESPAsyncWebServer.h#L60) at  `#define WEBSERVER_H`, `ESPAsyncWebServer` shouldn't use these constants, but does use them [WebRequest.cpp#L254](https://github.com/me-no-dev/ESPAsyncWebServer/blob/f6fff3f91ebf45b91ca4cff2460d2febd61f1e27/src/WebRequest.cpp#L254) on parse request.
```c++
#ifndef WEBSERVER_H
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111,
} WebRequestMethod;
#endif
```

jpcaware, 2021-03-01:
Hello
a big THANK  to egrekov
#define WEBSERVER_H
before import ESPAsyncWebServer lib
#include "ESPAsyncWebServer.h"

fixed my issue
Regards,

Benvorth, 2021-12-13:
> ### [[image]](/egrekov) ** [egrekov](/egrekov)**

Instead of just defining `WEBSERVER_H` I had to include the whole "Webserver.h" to resolve my issuess:

`// #define WEBSERVER_H
#include "WebServer.h"
#include <ESPAsyncWebServer.h>`

MaralS, 2022-04-28:
@egrekov Oh god you're my life saver ! I just opened an issue and it cost me two hours of struggle to finally see the solution. Thanks a lot. It would be great to update the library documentation to prevent these kind of issues.

AdrienV, 2023-09-24:
Defining WiFiManager.h at the top of my file solved the problem for me

```
#include "WiFiManager.h"
WiFiManager wifiManager;

#define WEBSERVER_H
#include "ESPAsyncWebServer.h" 

// My other libs used
#include <PubSubClient.h>
#include <WiFiClientSecure.h>
```

zixzelz, 2024-01-14:
Hello everyone,

Thank you for the provided solutions.

Regrettably, I have encountered situations where these solutions do not work. An example is when utilizing the "HTTP_PUT" or "HTTP_DELETE" methods. These methods have different values; for instance, in "WebServer" (esp32), "HTTP_PUT" has a value of 4, whereas in "ESPAsyncWebServer," it has a value of 0b00001000 (8). Similarly, "HTTP_DELETE" in "WebServer" has a value of 0, but in "ESPAsyncWebServer," it is 0b00000100 (4). On the other hand, the "HTTP_GET" method has a consistent value of 1 for both "WebServer" and "ESPAsyncWebServer."

This discrepancy leads to an issue where registering two functions for "HTTP_DELETE" and "HTTP_PUT" methods may result in unexpected behavior. If a request with the "PUT" method is made from the browser, the application will call the function registered for the "HTTP_DELETE" method. This occurs because our application uses methods from "WebServer", and the internal logic of "ESPAsyncWebServer" lacks the definition of "WEBSERVER_H," causing it to use the original values from "WebRequestMethod" enum in "ESPAsyncWebServer.h".

Has anyone discovered a workaround to address this issue?

Best Regards,