me-no-dev/ESPAsyncWebServer #419: Heap corruption under load

davepl, 2018-10-12:
I have found that if I POST data in a loop to the server it will heap-corrupt itself within half a dozen calls.

You should be able to repro by using POSTMAN and setting up a page to accept posted data, like so:

```
_server.on("/setcolor", HTTP_POST, 
		  	[this](AsyncWebServerRequest * pRequest)
      		{
				pRequest->send(200, "text/plain", "Data Received OK");				  
      		},
      		NULL,
      		[this](AsyncWebServerRequest *pRequest, uint8_t *data, size_t len, size_t index, size_t total)
      		{
				  pRequest->send(200, "text/plain", "Data Received OK");		
				  return;
                 }
```
I found that without data (the first case) it's fine, but if you send binary data with python or postman in a loop, it corrupts the heap right away.  This is true whether it can send all of the data in a single post or if it has to break it up, doesn't matter.

Is this any kind of known issue, or is there a potential workaround?  I'm going to have to try a GET with the data on the querystring for now, but that's horrible...

davepl, 2018-10-13:
BTW I was able to trigger this with a simple python script repeatedly sending a small data array to the POST interface (third parameter).

GeorgeFlorian, 2019-02-28:
Did you find any fix to this ?

davepl, 2019-02-28:
No, I replaced WebSockets with sockets and changed my REST API to a socket stream.  So I went old tech, which all works.  I wish the server was more reliable, but it looks pretty complicated internally!

Cheers,
Dave

closed 2019-10-05 by stale[bot]