me-no-dev/ESPAsyncWebServer #421: Includes for templates

dfelsch, 2018-10-21:
Hello,

Is there an easy way to include templates in other templates? My use-case is that I have a couple of pages to serve with ESPAsyncWebServer, but all of them should use a common header and footer.

Imagine my template in PROGMEM looks like this: 
`%HEADER% <p>Dynamic Content: %CONTENT%</p> %FOOTER%`

The %HEADER% template variable should be replaced by this (also in PROGMEM): 
`<html><head><title>%PAGETITLE%<title><head><body>`

Of course, %PAGETITLE% is again to be replaced dynamically. Is there an easy way to achieve this?

Thanks!

me21, 2018-11-01:
Doesn't it just work? I.e. if you replace %HEADER% template variable, isn't %PAGETITLE% replaced too?

dfelsch, 2018-11-02:
It's not like that %PAGETITLE% is still in the output. But I remember several hundred characters from the replacement string were missing in the output when playing around with it. Probably I hit a problem with the buffering.

Anyway, meanwhile I completely abandoned the template processing for my project. I replaced it by [response printing](https://github.com/me-no-dev/ESPAsyncWebServer#print-to-response) with `printf_P`:

```
const char html_header[] PROGMEM = "<html><head><title>%s<title><head><body>";
const char html_content[] PROGMEM = "<p>Dynamic Content: %s</p>";
const char html_footer[] PROGMEM = "</body></hmtl>";

AsyncResponseStream *response = request->beginResponseStream("text/html");
response->printf_P(html_header, "Fancy Page");
response->printf_P(html_content, "Stuff");
response->print(FPSTR(html_footer));
request->send(response);
```

For me, that's a viable solution, so I'm closing this issue.

closed 2018-11-02 by dfelsch