me-no-dev/ESPAsyncWebServer #425: How to slow down a file upload?

StarsoftAnalysis, 2018-10-29:
I'm trying to use ESPAsyncWebServer for a v-plotter.

I upload a large (megabytes) file of commands that control the stepper motors, and they run slowly so I need to put in many short `delay()`s.  

Will I need to create my own flow-control system, so that the Javascript in the web front-end only sends the data in small chunks, or is there a simpler way of slowing down the upload process?

Cheers,

Chris

me21, 2018-11-01:
The web server knows nothing about a v-plotter, so it's not a problem of the web server. ESP32/ESP8266 cannot hold the entire file of several megabytes in memory at once, so you'll need to send the data in chunks anyway.
I would implement it on Websockets, so that you send a chunk of commands, then you get a reply when these commands have been executed. Now you may send the next chunk if there is one.

penfold42, 2018-11-01:
Maybe add an SD card and write the file there temporarily and then process it when complete ?

StarsoftAnalysis, 2018-11-01:
Thanks me21.  I was previously using [esp8266webserver](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer), which had the convenient feature of being happy to upload a 10MB file however slowly the ESP8266 processed each chunk of data without timing out.  But it only handles a single connection, so the upload blocked everything else.  You're right -- with ESPAsyncWebServer I'll have to send the data a chunk at a time, either via ajax calls or a websocket.

And thanks penfold42 -- an SD card reader is a possible future upgrade.

me21, 2018-11-01:
I guess the upload was synchronous in that case, so it didn't accept the next chunk until the previous one was processed. Well, it just isn't working the same way in async world :)

StarsoftAnalysis, 2018-11-02:
I tried a hacky way of delaying the response via a timer: it worked for a while, but then threw an Exception(9) -- probably ran out of memory because it couldn't deallocate the request pointers.

So websockets are the next thing to try.

closed 2019-10-05 by stale[bot]