me-no-dev/ESPAsyncWebServer #429: ESP32  | Content-Encoding: gzip | Bootstrap

LuckyLuzz, 2018-11-12:
Hey,
nice lib! Took me a bit to understand alle the functions and unforgently I still don't. I couldn't get my head around serving coded gzip files. 
I have build a nice bootstrap page that has 860kb if zipped, the problem is that I can't get Mozilla Firefox to unzip the files and show them properly. 
My guess is that I am doing somethong wrong with the headers?  

HTML Page
    
    <link href="./css/bootstrap.min.css.gz" rel="stylesheet" media="screen">
    <link href="./css/fontawesome.css.gz" rel="stylesheet">
    <link href="./css/costum.css" rel="stylesheet">
    <script src="./js/jquery-3.3.1.slim.min.js.gz" crossorigin="anonymous"></script>
    <script src="./js/popper.min.js.gz" crossorigin="anonymous"></script>
    <script src="./js/bootstrap.min.js.gz" crossorigin="anonymous"></script>
    <script src="./js/fontawesome.js.gz" crossorigin="anonymous"></script>
    <script src="./js/asyncdata.js" crossorigin="anonymous"></script>

ESP32_Code
   server.on("", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");

That funktion serves the not compressed files but not the gzip files. 

Any ideas?

Pablo2048, 2018-11-12:
Don't put ./css/bootstrap.min.css.gz in your HTML page - just use ./css/bootstrap.min.css and Webserver automagically serve .gz file instead ;-) ...

LuckyLuzz, 2018-11-13:
Dude thank you! Well that was easy^^

closed 2018-11-13 by LuckyLuzz