me-no-dev/ESPAsyncWebServer #431: ESP32: Receiving Binary from JS over websocket

LuckyLuzz, 2018-11-16:
Hey,
in the ESP_AsyncFSBrowser.ino example there is the `(info->opcode == WS_TEXT)` methode to check if it is a text or a binary. As I send a string from my client to the websocketserver it does register it as text. But the other way around it does not. As I use in JS `parseInt()` to convert the string to integer it does not registier it as binary. 

 ```
else if (type == WS_EVT_DATA)
  {
    AwsFrameInfo *info = (AwsFrameInfo *)arg;
    String msg = "";
    if (info->final && info->index == 0 && info->len == len)
    {
      //the whole message is in a single frame and we got all of it's data
      Serial.printf("ws[%s][%u] %s-message[%llu]: ", server->url(), client->id(), (info->opcode == WS_TEXT) ? "text" : "binary", info->len);

      if (info->opcode == WS_TEXT)
      {
        for (size_t i = 0; i < info->len; i++)
        {
          msg += (char)data[i];
        }
      }
      else
      {
        char buff[3];
        for (size_t i = 0; i < info->len; i++)
        {
          sprintf(buff, "%02x ", (uint8_t)data[i]);
          msg += buff;
        }
      }
      Serial.printf("%s\n", msg.c_str());
      if (info->opcode == WS_TEXT)
        client->text("I got your text message");
      else
        client->binary("I got your binary message");
```
with JS Script
```
function SendData() {
  var textToSend = document.getElementById("inputText").value;
  ws.send(textToSend);
}
```
it does work, but with JS Script

```
function SendData() {
  var numberToSend = parseInt(document.getElementById("inputText").value);
  ws.send(numberToSend);
}
```
it does not. At least I don't get the BrowserConsole Message. As I send a text via integer I get "ws[/index][1] text-message[3]: NaN" in serial console, numbers are shown correct as expectet, but the console message itself it wrong because it should say "binary-message" instead of "text-message" and also it does not send the ""I got your binary message" to the client.

So what did I do wrong?

StarsoftAnalysis, 2018-11-23:
I don't think a single number counts as 'binary data' for this purpose.  You probably need to send a blob or a typed array.  See https://developer.mozilla.org/en-US/docs/Web/API/WebSocket/send

LuckyLuzz, 2018-11-26:
Hey,
thanks for you answer. And yes you are right. As I send:   

```
var arr2 = new ArrayBuffer(32);
  var view = new Float32Array(arr2);
  for (var i = 0; i < 8; i++)
    view[i] = i *17.5;
  ws.send(arr2);
```
from JS, the lib recognize it as binary. So rightnow I am working on a example for data transfer with websocket and different data typs in one array for Client and Serverside.

closed 2019-10-05 by stale[bot]