me-no-dev/ESPAsyncWebServer #432: Set-cookie header problems

kamil2234, 2018-11-18:
Hi

I try to set cookie with:

- expiration date, 
- max age
-  another path, 
- own domain, 
- secure, 
- HttpOnly

like:

`Set-Cookie: <cookie-name>=<cookie-value>; Domain=<domain-value>; Secure; HttpOnly`
or 
`Set-Cookie: qwerty=219ffwef9w0f; Domain=somecompany.co.uk; Path=/; Expires=Wed, 30 Aug 2019 00:00:00 GMT
`
How can I set such a cookie with this library?  My code is below:

```
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>

#include <Hash.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

String ESPSESSIONID = "NODATA";

AsyncWebServer server(80);

const char* ssid = "blah";
const char* password = "blahblah";

const char* PARAM_MESSAGE = "message";

String GetSession(){
       int bla = random(1111000, 99999999);   
      String esper = String(bla); 
      Serial.print("test sha1 int to string radnom: ");
      Serial.println(esper);
     
      String ESPSESSIONID = sha1(esper); 
      return ESPSESSIONID;
} /* END function GetSession */

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

void setup() {
   Serial.begin(9600);
   delay(1000);
  
   Serial.println("esp starts ");
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    Serial.println("");
  
    // Wait for connection
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
  
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){

         
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Hello World!");
    response->addHeader("Server","ESP Async Web Server");
    ESPSESSIONID = GetSession();
    String CookieSessionHeader = "ESPSESSIONID="+ESPSESSIONID;
    response->addHeader("Cache-Control","no-cache");
    response->addHeader("Set-Cookie", CookieSessionHeader);
         request->send(response);
    });

    server.onNotFound(notFound);

    server.begin();
}

void loop() {
}
```

StarsoftAnalysis, 2018-11-23:
Does this not work? 

`response->addHeader("Set-Cookie",  "id=a3fWa; Expires=Wed, 21 Oct 2015 07:28:00 GMT; Secure; HttpOnly");`

kamil2234, 2018-11-24:
sorry but no :(

closed 2019-10-05 by stale[bot]