me-no-dev/ESPAsyncWebServer #433: Stop / Start question

MartinDavidWaller, 2018-11-20:
Hi,

I have a project where I want to use the ESP32 as an access point and have the user enter Wi-Fi connection details via some HTML pages served up using your server. Once I have these details I plan to close the access point and connect directly to the users Wi-Fi where again I'd like to server up different HTML files.

I was wondering what is the best way of stopping / restarting the server while changing operating modes?

Many thanks

Martin

StarsoftAnalysis, 2018-11-23:
I've done something similar, and didn't need to restart the server when switching the WiFi between STA and AP modes.

[The documentation](https://github.com/me-no-dev/ESPAsyncWebServer#using-filters) shows how to serve different pages in the different modes.

MartinDavidWaller, 2018-11-26:
Excellent new. Many thanks for commenting.

closed 2019-10-05 by stale[bot]