me-no-dev/ESPAsyncWebServer #435: Connection problems when using mobile devices

sirluckyluke, 2018-11-22:
Hello folks,

I am working on a project where I try to control servos over a web page. For this, the esp32 is set up in hotspot mode with a DNS server for name resolution and the async web server for content delivery. 

Everything is working fine on my development machine (Ubuntu/Firefox/Chrome), I can connect to the esp32 and the web page is shown correctly and also a page refresh is possible. But when I try the same with my mobile phone (android 9) or my tablet (iPad / latest OS) it worked on the first try but when I refresh the page it gets stuck in the refreshing routine and nothing more happens. The weird thing is when I then try to connect from my dev machine the network manager shows an authentication error and wants me to type in the password again (which is not working then).  After a hard reset, it is working in most cases till the page refreshing. 
Another observation is, that the mobile devices are not really willing to connect to hotspots which are not providing internet. Sometimes my android phone and tablet are disconnecting themselves from the wifi because of no internet or they start using the cellular connection. Furthermore, I see on the webserver that the devices requesting some files like "hotspot-detect.html" or "generate_204". 

Could this be some problems with the captive portal detection of mobile devices? Which need some special response from the web server? Or a special authentification routine?

I hope someone could get me some advice on how to deal with this.

sirluckyluke, 2018-11-22:
if forgot to mention:
I using the esp32 dev module with the latest files for the async web server and core files.

deniss1711, 2018-11-24:
Using this implementation https://github.com/boblemaire/ESPAsyncSDWebServer of the server having similar issue. Chrome/Firefox/Edge all do a correct reload of the page. When page is loaded heap is back to start size.

First load of page on mobile devices (iOS/chrome) is ok when app was closed before. A reload of page ends in a stuck page and heap size is not recovering from this. ESP is not reachable from normal computer at this stage.

deniss1711, 2019-02-25:
@sirluckyluke Sorry no solution found yet. But my webserver is not sending any data after calling a large js page from the server on a chrome mobile. On all Windows browsers all is working ok.

hjanshu, 2019-04-23:
I'm facing the same issue. I'm hosting html and js files on the on-board spiffs. The page loads without issues on windows machine but on Mobile device the page is stuck on loading.

closed 2019-10-05 by stale[bot]

aleuarore, 2021-03-26:
Hi guys, 
did you find a solution? I'm having this same problem.

zekageri, 2021-03-27:
Chrome and other browsers are requesting some additional default things from the server after the page load such as favicon.ico or manifest.json. i found out that if i serve these files too the problem has gone.

openedhardware, 2021-11-24:
@zekageri 
What files did you add to get it working?

I manually added `favicon.ico` and `manifest.json` but still not working..

Here is my folder structure:
[image]

And I am serving with the following code:
```c++
server.serveStatic("/", SPIFFS, "/www/").setDefaultFile("index.html");
```

Thanks!

zekageri, 2021-11-26:
Do you have problems on just AP mode?
What's the decoded error you get?

openedhardware, 2021-11-26:
@zekageri 

I am just using AP mode all the time.

And how can I check the decoded error on the mobile browser?

zekageri, 2021-11-26:
Use the esp serial monitor to check the error message when it reboots

zekageri, 2021-11-26:
Or if it is not rebooting then serial print some line when you serve the page to determine where it stuck

openedhardware, 2021-11-26:
There is no error message... 192.168.4.1(default IP address of the AP) doesn't reply at all on my android phone.

But it works on my laptop with wifi dongle...

So strange..

openedhardware, 2021-11-26:
@zekageri 

I have just created a new repository to test this issue - https://github.com/openedhardware/TestESP32AsyncWebServer

This is really simple firmware, but the webserver is not working on mobile phones...

zekageri, 2021-11-26:
First of all. Replace SPIFFS with LITTLEFS. SPIFFS is deprecated and LITTLEFS is much faster and reliable.

Try it without static file serving and create an endpoint in the setup beforfe the `server.begin()` like this:

```C
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    AsyncWebServerResponse* response = request->beginResponse(SPIFFS, "/index.html", "text/html");
    request->send(response);
});
```

If it is working, something wrong with the path's on the static serving. 
If it is not working like this, tell me.

**You should define your page endpoints like this and serve the static content like images and fonts like this:**

`server.serveStatic("/", LITTLEFS, "/");`

**And handle the not found endpoint like this:**

```C
server.onNotFound([](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/notFound.html", "text/html");
    request->send(response);
});
```

Other than that actually i can't see any problem with your code.

openedhardware, 2021-11-26:
@zekageri 

```c
  WiFi.disconnect();
  delay(50);
  WiFi.mode(WIFI_AP); // Access Point mode
  delay(50);
  WiFi.softAP(ssid, password);
  delay(50);
```

This resolves the issue! THANKS, A MILLION! 👍

openedhardware, 2022-01-18:
Hi, @zekageri 

I am now using the same code on NodeMCU, but doesn't work at all.

Any idea to resolve this issue? Seems power issue?

Thanks again!

ddp75, 2023-03-18:
I had a similar problem (http_response_code_failure) using : 

`if(!request->authenticate(my_username,` my_password)){return request->requestAuthentication();}

Simply remove this line to solve the problem... If it helps?
I'm looking for another solution to protect access to the page... :)

Airwolf1979, 2023-04-23:
android phone did not want to connect, fixed by adding the port 80, 192.168.4.1:80

ktrzeciaknubisa, 2023-12-28:
In my case ESP32 module was constantly restarting when I was accessing the web server via mobile device (android)

My setup was simple: 

` server.serveStatic("/", LittleFS, "/").setDefaultFile("/index.html");`

but apparently I lacked the `onNotFound()` method, as @zekageri mentioned.
Once I added it, my ESP stopped restarting when accessing from my phone.

```
server.onNotFound([](AsyncWebServerRequest *request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/notFound.html", "text/html");
    request->send(response);
});
```

ktrzeciaknubisa, 2023-12-28:
Actually it did not help - I was too early. 
It worked couple of times, but then it broke again (Arduino crashes when accessing the web server from android mobile)
.