me-no-dev/ESPAsyncWebServer #445: ESPAsyncWebServer Compilation Errors

CubanJew, 2018-12-18:
Including <ESPAsyncWebServer.h> causes all sorts of compilation redeclaration errors with PlatformIO for ESP8266.

```

^
In file included from src\main.h:3:0,
from src\main.cpp:2:
C:\Users\***\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:29: note: previous declaration 'HTTPMethod HTTP_GET'
enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
^
In file included from src\main.h:17:0,
from src\main.cpp:2:

.piolibdeps\ESP Async WebServer_ID306\src/ESPAsyncWebServer.h:57:18: error: redeclaration of 'HTTP_POST'
HTTP_POST    = 0b00000010,
^
In file included from src\main.h:3:0,
from src\main.cpp:2:
C:\Users\***\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:39: note: previous declaration 'HTTPMethod HTTP_POST'
enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
^
In file included from src\main.h:17:0,
from src\main.cpp:2:
.piolibdeps\ESP Async WebServer_ID306\src/ESPAsyncWebServer.h:58:18: error: redeclaration of 'HTTP_DELETE'
HTTP_DELETE  = 0b00000100,
^
In file included from src\main.h:3:0,
from src\main.cpp:2:
C:\Users\***\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:31:72: note: previous declaration 'HTTPMethod HTTP_DELETE'
enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

```

closed 2019-10-05 by stale[bot]

alexhopeoconnor, 2022-05-21:
Yeah I am getting this error also in a new project I setup.

ramgo0, 2022-06-20:
Same issue, still unresolved.