me-no-dev/ESPAsyncWebServer #447: ESPAsyncWebServer not working with ESP8266IRremote

wuhanstudio, 2018-12-20:
Hello, 

I'm trying to build an infrared remote controller using ESP8266 with ESPAsyncWebServer and IRremoteESP8266, but it seems there are some sorts of conflicts between these two libraries.

At first, I tried to send with:

` ac.setPower(true);  `
`ac.send()`

It worked before and after invoking `server.begin()`, but not inside the handler `server.on("/", HTTP_GET, [](AsyncWebServerRequest *request)`

At first, I thought it might relate to interrupts, so I added: 

`noInterrupt();`
`ac.send();`
`interrupt();`

But it is still not working. Is there any way to use IRremoteESP8266 within a hander?

Thanks a lot.

Related library: [https://github.com/markszabo/IRremoteESP8266](https://github.com/markszabo/IRremoteESP8266)

penfold42, 2018-12-20:
Probably not - you can’t block or use delay() in the handler.

Better to set a global variable or add to a queue and then have a loop() handler that checks this and does the actual work of sending

hreintke, 2018-12-24:
You can use the schedule_function.
Scheduled function are executed after each loop iteration.

Should be something like 
`schedule_function([](){ac.sent();})`

wuhanstudio, 2018-12-24:
Finally, I tried native ESP8266WebServer and it worked.

> https://github.com/wuhanstudio/ESP8266-Infrared

closed 2019-10-05 by stale[bot]