me-no-dev/ESPAsyncWebServer #450: Error with gzip'd static files

jnicolson, 2018-12-26:
This issue had me scratching my head for hours.  A route setup like:

`server.on("/404.html", HTTP_GET, [](AsyncWebServerRequest *request) {
      AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/404.html");      
      response->addHeader("Server", "ESP32");
      request->send(response);
});`

would get a HTTP status code 200 and you could see the custom header, so it was definitely this handler handling the request.  However the file size was 0 and the Content-Encoding header wasn't sent.

Investigating lead me to https://github.com/me-no-dev/ESPAsyncWebServer/blob/95dedf7a2df5a0d0ab01725baaacb4f982dedcb2/src/WebResponses.cpp#L508

which lead me to espressif/arduino-esp32#1264 . Upstream bug but thought i'd document in here because this was the first place I looked when troubleshooting!

remoteme, 2019-07-06:
This also happens for me
`fs.exists(_path) ` returns true if `_path+.gz` exists
is it possible to make some workaround the simplest is to replace line
` if(!download && !fs.exists(_path) && fs.exists(_path+".gz")){ ` 
with
 ` if(!download && fs.exists(_path+".gz")){`
 so webserver prefers gz files if exists ?

atanisoft, 2019-07-06:
@remoteme have you tried with arduino-esp32 1.0.3-rc1?

closed 2019-10-05 by stale[bot]