me-no-dev/ESPAsyncWebServer #451: Crash when delay is evoked in handler

treii28, 2018-12-27:
I've been trying to trace a problem and it seems to only trigger if I have any kind of 'delay(n)' call anywhere in any of the functions called by my handler. I'm not entirely sure how to read the stack, but the results show tcp/ip4 and mem_malloc. I've traced the lines mentioned and all the files are in lwip2 so I'm assuming it's something related to the web call, but as i say, it only seems to trigger when any 'delay()' is run.

```
0x4021ab4c: tcp_input at core/tcp_in.c line 435
0x4021f224: mem_malloc at core/mem.c line 136
0x4021e9ad: ip4_input at core/ipv4/ip4.c line 685
0x402180b8: ethernet_input_LWIP2 at netif/ethernet.c line 182
0x40217ef3: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 439
0x4023ad02: ethernet_input at glue-esp/lwip-esp.c line 363
0x4023ad13: ethernet_input at glue-esp/lwip-esp.c line 371
```

My code 
[https://github.com/treii28/esp01cam](url)

I'm trying to get a jpeg camera working on an esp01 with 1m

I traced it down to the delay after adding debug lines. Initially I was dropping print commands at various points followed by a delay. It would never get past the first one. I eventually, through trial and error, removed the delay and would find it would crash at any delay in the functions themselves. Removing those, I was able to get it to go all the way through.

Pablo2048, 2018-12-27:
Did you read this https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember ? Especially the "You can not use yield or delay or any function that uses them inside the callbacks" one...

treii28, 2018-12-27:
Nope, thanks!

Pablo2048, 2018-12-27:
:-) Ok, so please close this issue...

closed 2019-10-05 by stale[bot]