me-no-dev/ESPAsyncWebServer #455: When using OTA Update @ ESP8266

ledsi, 2019-01-03:
Hi, today i rewrote one of my ESP8266 projects to use with ESP32.
After a few complications and errors its just working fine and the update was less pain then i thought.

But... i had to skip my OTA Update because it came up with an error.
Is there any other way to user OTA on ESP32?
My program ist just working fine on ESP8266 and the update progress is the same like hier in code examples.

I get these errors:
```
error: 'Update' was not declared in this scope
Update.runAsync(true);
error: 'class EspClass' has no member named 'getFreeSketchSpace'
```

atanisoft, 2019-01-04:
or even: https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino#L138

which does what you are trying to do...

lbernstone, 2019-01-05:
You can just remark the offending lines, which are ESP8266 specific.  ESP32 is natively async.

ledsi, 2019-01-06:
Thanks for your reply.

Now i just got it work, removing the lines
`Update.runAsync(true);`
and 
`(ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000)`

But i think, without checking free sketch space it might crash if my file is to big or?

zuqualla, 2019-06-05:
> You aren't including the Update module in your code and with the ESP32 there is need to pass parameters to Update.begin(), you can see an example here that will work with a few tweaks for ESPAsync: https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/WebUpdate/WebUpdate.ino

Did you make a typo and  mean there is **no** need to pass arguments? cause the exemple you refer to has no arguments for  Update.begin() ...?

atanisoft, 2019-06-06:
@zuqualla most likely yes

closed 2019-10-05 by stale[bot]