me-no-dev/ESPAsyncWebServer #461: Corrupted websocket data + reboot using example code

tripplet, 2019-01-08:
Hi,

### Problem

The recevied websocket data by the ESP seems to be corrupted when it is send in multiple frames by the browser.
There seems to be same "e" at the end of frames which does not belong there and the data is corrupted in the 3. frame. After that the ESP reboots.

### Steps to reproduce

- I used the file server example from this repository [ESP_AsyncFSBrowser.ino](https://github.com/me-no-dev/ESPAsyncWebServer/blob/a7c4dfb04fd6a06001d9f87ba9935cc67cb256d5/examples/ESP_AsyncFSBrowser/ESP_AsyncFSBrowser.ino) with PlatformIO.
- No modification only the wifi name and password were changed.
- platformio.ini file: 
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
upload_speed = 921600
monitor_speed = 115200
lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git
```
- Wait for the nodemcu to connect to the wifi and browse to the IP
  http://<i></i>192.168.0.113/heap

- Using the chrome dev tools executed the following JavaScript code:
```javascript
var ws = new WebSocket("ws://" + location.host + "/ws");
ws.onmessage = function(evt) { console.log(evt.data) };
```
- Wait for the `Hello Client 1 :)` response
- Execute the JavaScript code:
```javascript
ws.send("123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890");
```

- Chrome gets the response `I got your text message`
- Serial debug prints from the nodemcu:
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with *****, channel 6
dhcp client start...
ip:192.168.0.113,mask:255.255.255.0,gw:192.168.0.1
ws[/ws][1] connect
ws[/ws][1] pong[0]:
ws[/ws][1] text-message start
ws[/ws][1] frame[0] start[lu]
ws[/ws][1] frame[0] text[lu - lu]: 123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678e
ws[/ws][1] frame[0] text[lu - lu]: 90123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234e
ws[/ws][1] frame[0] text[lu - lu]: 5678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890���␄y␁�␅34567890123456789012345678901234567890123456789012345678901234567890123456789012345
ws[/ws][1] frame[0] end[lu]
ws[/ws][1] text-message end
Fatal exception 28(LoadProhibitedCause):
epc1=0x402209a8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x05d004c4, depc=0x00000000

Exception (28):
epc1=0x402209a8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x05d004c4 depc=0x00000000

ctx: sys
sp: 3fffece0 end: 3fffffb0 offset: 01a0

>>>stack>>>
LONG STACK TRACE....
```
Followed by a reboot

Pako2, 2019-01-08:
See [#457](https://github.com/me-no-dev/ESPAsyncWebServer/issues/457).

tripplet, 2019-01-08:
Thanks that fixes it.

Pako2, 2019-01-08:
You're welcome. I am glad my work is useful for others as well.
It is strange that this "example sketch" is about one year old and no one noticed that bug.
And now, during the four days we were suddenly two users who have encountered the same problem ...