me-no-dev/ESPAsyncWebServer #464: ESPAsyncWebServer wont compile

Dr-Zoidberg-de, 2019-01-16:
Ive read about this error, and it beeing fixed via updates but i cant compile regardless.
Im using the latest arduino ide and esp32core.
Is there something im missing?
i installed the latest asyncwebserver and asynctcp, boards, ide, wifi.h

> Arduino: 1.8.8 (Windows 10), Board: "WEMOS LOLIN32, 80MHz, 921600"

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:32: error: field 'call' has incomplete type 'tcpip_api_call_data'

     struct tcpip_api_call_data call;

                                ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:12: note: forward declaration of 'struct tcpip_api_call_data'

     struct tcpip_api_call_data call;

            ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_output(tcp_pcb*)':

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:295:70: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_output_api, (struct tcpip_api_call_data*)&msg);

                                                                      ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:295:70: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_write(tcp_pcb*, const char*, size_t, uint8_t)':

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:311:69: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_write_api, (struct tcpip_api_call_data*)&msg);

                                                                     ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:311:69: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_recved(tcp_pcb*, size_t)':

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:326:70: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]

     tcpip_api_call(_tcp_recved_api, (struct tcpip_api_call_data*)&msg);

                                                                      ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:326:70: error: cannot convert 'tcpip_api_call_data*' to 'tcpip_api_call*' for argument '2' to 'err_t tcpip_api_call(tcpip_api_call_fn, tcpip_api_call*)'

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:259:12: note: class type 'tcpip_api_call_data' is incomplete

     struct tcpip_api_call_data call;

            ^

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In function 'esp_err_t _tcp_connect(tcp_pcb*, ip_addr_t*, uint16_t, tcp_connected_fn)':

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:342:71: error: invalid conversion from 'err_t (*)(tcpip_api_call_data*) {aka signed char (*)(tcpip_api_call_data*)}' to 'tcpip_api_call_fn {aka signed char (*)(tcpip_api_call*)}' [-fpermissive]
[... 77 lines cut ...]
C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp: In member function 'void AsyncClient::_dns_found(ip_addr*)':

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:643:33: error: invalid use of incomplete type 'struct ip_addr'

         connect(IPAddress(ipaddr->u_addr.ip4.addr), _connect_port);

                                 ^

In file included from C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp:24:0:

C:\Users\drzoi\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.h:46:8: note: forward declaration of 'struct ip_addr'

 struct ip_addr;

        ^

Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\drzoi\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Fehler beim Kompilieren für das Board WEMOS LOLIN32.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

atanisoft, 2019-01-16:
Use AsyncTCP 1.0.3 or later.

Dr-Zoidberg-de, 2019-01-16:
i am as i stated in the first 3 lines

atanisoft, 2019-01-16:
update your arduino-esp32 core to the 1.0.1 release if you haven't already. AsyncTCP was updated recently to reflect an update in the arduino-esp32 core code.

Dr-Zoidberg-de, 2019-01-16:
again, everything is up to date. sorry if i sound ungrateful, it's just that i mentioned it twice in the initial post.

atanisoft, 2019-01-16:
Something isn't right in the arduino IDE then, it is not using the updated bits that is certain. It is mixing and matching old and new. You aren't the first to report this sort of oddness with the arduino ide (it isn't a problem with either arduino-esp32 or asynctcp/espasyncwebserver).

MileBuurmeijer, 2019-01-20:
Any results @Dr-Zoidberg-de ?
I am having the same issues and are a kind of lost what to do about it.

Dr-Zoidberg-de, 2019-01-20:
sadly not. i havent answered because im planning on e fresh windows install since i cant rule out traces of old code, but im lacking spare time at the moment

wieb18, 2019-01-20:
I am also having the same issues. 

Using ESP32 Dev board, Windows 10, and Arduino IDE 1.8.8 with arduino-esp32 core version 1.0.0. Can't compile ESPAsyncWebServer and AsyncTCP, Arduino IDE showing tons of errors.

Did not use newest arduino-esp32 core version 1.0.1 because something not right on the newest version. Read the issue here: [https://github.com/pbecchi/ESP32_ping/issues/8](https://github.com/pbecchi/ESP32_ping/issues/8) and [http://forum.arduino.cc/index.php?topic=591282.new;topicseen#new](http://forum.arduino.cc/index.php?topic=591282.new;topicseen#new)

Is there any chance for us to use ESPAsyncWebServer with arduino-esp32 core version 1.0.0?

atanisoft, 2019-01-20:
@wieb18 use AsyncTCP@1.0.0 and it should work fine for you with the older adruino-esp32 core.

wieb18, 2019-01-20:
@atanisoft - Thanks for very quick reply.

But where I could find and download AsyncTCP@1.0.0?

In this Github page: [https://github.com/me-no-dev/AsyncTCP](https://github.com/me-no-dev/AsyncTCP) is AsyncTCP@1.0.3

Please guide me...

atanisoft, 2019-01-20:
@wieb18 you can download a snapshot from this commit: https://github.com/me-no-dev/AsyncTCP/tree/952b7eb6ba62f7071f7da2a274d36e06b97de572 Click on the green Clone or Download link on the right side.

wieb18, 2019-01-20:
Clone from [https://github.com/me-no-dev/AsyncTCP/tree/952b7eb6ba62f7071f7da2a274d36e06b97de572](https://github.com/me-no-dev/AsyncTCP/tree/952b7eb6ba62f7071f7da2a274d36e06b97de572)

According library.json file, its AsyncTCP@1.0.2

Now.... no error on Arduino IDE Compiler, ESPAsyncWebServer@1.2.0 with AsyncTCP@1.0.2 working fine.

Thanks a lot @atanisoft for the guidance.

atanisoft, 2019-01-20:
@wieb18 you're welcome, the version was 1.0.0 and 1.0.2 is just before the merge of the IDF update branch which contains the breaking changes you are seeing with the older arduino-esp32 core.

alonsoelectronica, 2019-02-05:
> @wieb18 you can download a snapshot from this commit: https://github.com/me-no-dev/AsyncTCP/tree/952b7eb6ba62f7071f7da2a274d36e06b97de572 Click on the green Clone or Download link on the right side.

Thanks @atanisoft you´re awesome!!

davepl, 2019-02-06:
I can confirm that I too was having the same build issue with 1.03 and it resolved using the 1.02 snapshot linked above!  Thanks all!

closed 2019-10-05 by stale[bot]

max5555, 2020-08-28:
```
lib_deps = 
    AsyncTCP@1.0.3
    ESPAsyncWebServer@1.2.0

```
And also
`lib_ignore=Time`
see https://github.com/me-no-dev/ESPAsyncWebServer/issues/60