me-no-dev/ESPAsyncWebServer #475: AsyncJson build error on PlatformIO, but works in Arduino IDE

dacodekid, 2019-02-18:
Soon you try to `#include <AsyncJson.h>`, the compilation broke in `platformio`.  It however compiles file in `Arduino IDE` though.

```
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino

lib_deps =
    ArduinoJson@~5.13.4    
    ESP Async WebServer
```

```
In file included from src/main.cpp:3:0:
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:68:55: error: expected class-name before '{' token
class AsyncJsonResponse: public AsyncAbstractResponse {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In constructor 'AsyncJsonResponse::AsyncJsonResponse(bool)':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:75:7: error: '_code' was not declared in this scope
_code = 200;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:76:7: error: '_contentType' was not declared in this scope
_contentType = JSON_MIMETYPE;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In member function 'size_t AsyncJsonResponse::setLength()':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:86:7: error: '_contentLength' was not declared in this scope
_contentLength = _root.measureLength();
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In member function 'size_t AsyncJsonResponse::_fillBuffer(uint8_t*, size_t)':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:94:29: error: '_sentLength' was not declared in this scope
ChunkPrint dest(data, _sentLength, len);
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: At global scope:
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:100:14: error: 'function' in namespace 'std' does not name a template type
typedef std::function<void(AsyncWebServerRequest *request, JsonVariant &json)> ArJsonRequestHandlerFunction;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:102:59: error: expected class-name before '{' token
class AsyncCallbackJsonWebHandler: public AsyncWebHandler {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:106:3: error: 'WebRequestMethodComposite' does not name a type
WebRequestMethodComposite _method;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:107:3: error: 'ArJsonRequestHandlerFunction' does not name a type
ArJsonRequestHandlerFunction _onRequest;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:111:50: error: 'ArJsonRequestHandlerFunction' has not been declared
AsyncCallbackJsonWebHandler(const String& uri, ArJsonRequestHandlerFunction onRequest) : _uri(uri), _method(HTTP_POST|HTTP_PUT|HTTP_PATCH), _onRequest(onRequest), _maxContentLength(16384) {}
^
In file included from src/main.cpp:3:0:
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:112:18: error: 'WebRequestMethodComposite' has not been declared
void setMethod(WebRequestMethodComposite method){ _method = method; }
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:114:18: error: 'ArJsonRequestHandlerFunction' has not been declared
void onRequest(ArJsonRequestHandlerFunction fn){ _onRequest = fn; }
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:116:26: error: 'AsyncWebServerRequest' has not been declared
virtual bool canHandle(AsyncWebServerRequest *request) override final{
Compiling .pioenvs/nodemcu-32s/libc16/WiFi/WiFiScan.cpp.o
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:133:30: error: 'AsyncWebServerRequest' has not been declared
virtual void handleRequest(AsyncWebServerRequest *request) override final {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:148:29: error: 'AsyncWebServerRequest' has not been declared
virtual void handleUpload(AsyncWebServerRequest *request, const String& filename, size_t index, uint8_t *data, size_t len, bool final) override final {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:150:27: error: 'AsyncWebServerRequest' has not been declared
virtual void handleBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) override final {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:116:16: error: 'virtual bool AsyncCallbackJsonWebHandler::canHandle(int*)' marked 'override', but does not override
[... 79 lines cut ...]
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In member function 'virtual void AsyncCallbackJsonWebHandler::handleBody(int*, uint8_t*, size_t, size_t, size_t)':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:151:9: error: '_onRequest' was not declared in this scope
if (_onRequest) {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:153:33: error: request for member '_tempObject' in '* request', which is of non-class type 'int'
if (total > 0 && request->_tempObject == NULL && total < _maxContentLength) {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:154:18: error: request for member '_tempObject' in '* request', which is of non-class type 'int'
request->_tempObject = malloc(total);
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:156:20: error: request for member '_tempObject' in '* request', which is of non-class type 'int'
if (request->_tempObject != NULL) {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:157:36: error: request for member '_tempObject' in '* request', which is of non-class type 'int'
memcpy((uint8_t*)(request->_tempObject) + index, data, len);
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In member function 'virtual bool AsyncCallbackJsonWebHandler::isRequestHandlerTrivial()':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:161:65: error: '_onRequest' was not declared in this scope
virtual bool isRequestHandlerTrivial() override final {return _onRequest ? false : true;}
^
Compiling .pioenvs/nodemcu-32s/libc16/WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/nodemcu-32s/lib99a/FS/FS.cpp.o
Compiling .pioenvs/nodemcu-32s/lib99a/FS/vfs_api.cpp.o
*** [.pioenvs/nodemcu-32s/src/main.cpp.o] Error 1
```

atanisoft, 2019-02-18:
Your lib_deps looks incorrectly specified for ArduinoJson, it should not have the ~ character.

dacodekid, 2019-02-18:
That's no the issue. This was just an artifact when  I tried out each version (from 6 beta -> 5.13.4) one by one.

atanisoft, 2019-02-18:
I use PIO successfully and have the following in my lib_deps:
```
lib_deps=
  SPI
  ArduinoJson@5.13.4
  AsyncTCP
  ESP Async WebServer
  ESP8266_SSD1306
  LiquidCrystal_PCF8574
  ESPmDNS
  Wire
  EEPROM
  Update
```
The ArduinoJson 6.x betas are not supported and even advised against usage by the author. If you specify it as I have done above it will pin you to that specific version, you may need to do a clean build (remove the .piolibdeps and .pioenvs folders).

dacodekid, 2019-02-18:
Here I copied exactly your libs.. and getting this result while compiling.

```
Processing nodemcu-32s (framework: arduino; platform: espressif32; board: nodemcu-32s)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'U`pdate'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'U`pdate'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <ArduinoJson> 5.13.4
|-- <AsyncTCP> 1.0.3
|-- <ESP Async WebServer> 1.2.0
|   |-- <AsyncTCP> 1.0.3
|   |-- <ArduinoJson> 5.13.4
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <LiquidCrystal_PCF8574> 1.1.0
|   |-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
|-- <Wire> 1.0.1
|-- <EEPROM> 1.0
Compiling .pioenvs/nodemcu-32s/src/main.cpp.o
Generating partitions .pioenvs/nodemcu-32s/partitions.bin
Compiling .pioenvs/nodemcu-32s/libc5d/SPI/SPI.cpp.o
Archiving .pioenvs/nodemcu-32s/lib380/libArduinoJson_ID64.a
Indexing .pioenvs/nodemcu-32s/lib380/libArduinoJson_ID64.a
Compiling .pioenvs/nodemcu-32s/libfc6/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/AsyncEventSource.cpp.o
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/AsyncWebSocket.cpp.o
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/SPIFFSEditor.cpp.o
Archiving .pioenvs/nodemcu-32s/libc5d/libSPI.a
Indexing .pioenvs/nodemcu-32s/libc5d/libSPI.a
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebAuthentication.cpp.o
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebHandlers.cpp.o
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebRequest.cpp.o
In file included from .piolibdeps/ESP Async WebServer_ID306/src/AsyncEventSource.h:29:0,
from .piolibdeps/ESP Async WebServer_ID306/src/AsyncEventSource.cpp:21:
.piolibdeps/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:27:16: fatal error: FS.h: No such file or directory

************************************************************
* Looking for FS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FS.h"
* Web  > https://platformio.org/lib/search?query=header:FS.h
*
************************************************************

compilation terminated.
*** [.pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/AsyncEventSource.cpp.o] Error 1
Compiling .pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebResponses.cpp.o
In file included from .piolibdeps/ESP Async WebServer_ID306/src/AsyncWebSocket.h:32:0,
from .piolibdeps/ESP Async WebServer_ID306/src/AsyncWebSocket.cpp:22:
.piolibdeps/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:27:16: fatal error: FS.h: No such file or directory

************************************************************
* Looking for FS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FS.h"
* Web  > https://platformio.org/lib/search?query=header:FS.h
*
[... 37 lines cut ...]
* CLI  > platformio lib search "header:FS.h"
* Web  > https://platformio.org/lib/search?query=header:FS.h
*
************************************************************

compilation terminated.
*** [.pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebRequest.cpp.o] Error 1
In file included from .piolibdeps/ESP Async WebServer_ID306/src/WebResponses.cpp:21:0:
.piolibdeps/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:27:16: fatal error: FS.h: No such file or directory

************************************************************
* Looking for FS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FS.h"
* Web  > https://platformio.org/lib/search?query=header:FS.h
*
************************************************************

compilation terminated.
*** [.pioenvs/nodemcu-32s/libfb8/ESP Async WebServer_ID306/WebResponses.cpp.o] Error 1
==================================================================== [ERROR] Took 3.42 seconds ====================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

atanisoft, 2019-02-18:
Add the following to your platformio.ini:
```
lib_compat_mode=2
lib_ldf_mode=chain+
```
these will configure the library dependency finder to only consider libraries that are supported for your target (esp32, arduino) and the chain+ will have it delve into the libraries to find their dependencies. The default settings (chain, 0) for the LDF may cause issues with the wrong libraries to be picked in a few cases (WiFi lib from Arduino/AVR picked up instead of the ESP32 built-in as example).

You can also add FS to lib_deps but with the two lines above it shouldn't be necessary.

dacodekid, 2019-02-18:
@atanisoft 

1. Adding those lines alone didn't solve the problem for me. So I tried your other suggestion: Adding `FS` to lib - which complained about `WiFi.h` dependency - then `SPI.h`.  Once I added them all, with below, it finally compiled.

```
lib_deps=
  ArduinoJson@5.13.4
  ESP Async WebServer
  FS
  WiFi
  SPI
```

However, once I `#include <AsyncJson.h>` it went back to the original error (first in this thread).

```
Processing nodemcu-32s (framework: arduino; platform: espressif32; board: nodemcu-32s)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=64 @ 5.13.4
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=306
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=870
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain+) COMPATIBILITY(strict)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.4
|-- <ESP Async WebServer> 1.2.0
|   |-- <AsyncTCP> 1.0.3
|   |-- <WiFi> 1.2.7
|   |   |-- <SPI> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <WiFi> 1.2.7
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <AsyncTCP> 1.0.3
Compiling .pioenvs/nodemcu-32s/src/main.cpp.o
Generating partitions .pioenvs/nodemcu-32s/partitions.bin
Archiving .pioenvs/nodemcu-32s/lib380/libArduinoJson_ID64.a
Indexing .pioenvs/nodemcu-32s/lib380/libArduinoJson_ID64.a
Compiling .pioenvs/nodemcu-32s/libfc6/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/nodemcu-32s/libc5d/SPI/SPI.cpp.o
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/WiFi.cpp.o
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/WiFiClient.cpp.o
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/WiFiServer.cpp.o
Archiving .pioenvs/nodemcu-32s/libc5d/libSPI.a
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/WiFiUdp.cpp.o
Indexing .pioenvs/nodemcu-32s/libc5d/libSPI.a
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/utility/server_drv.cpp.o
Compiling .pioenvs/nodemcu-32s/lib824/WiFi_ID870/utility/spi_drv.cpp.o
In file included from src/main.cpp:2:0:
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:68:55: error: expected class-name before '{' token
class AsyncJsonResponse: public AsyncAbstractResponse {
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h: In constructor 'AsyncJsonResponse::AsyncJsonResponse(bool)':
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:75:7: error: '_code' was not declared in this scope
_code = 200;
^
.piolibdeps/ESP Async WebServer_ID306/src/AsyncJson.h:76:7: error: '_contentType' was not declared in this scope
_contentType = JSON_MIMETYPE;
^
```

atanisoft, 2019-02-18:
There is a missing class type declaration in AsyncJson.h for sure, you will need to make sure you are including ESPAsyncWebServer.h *BEFORE* AsynJson.h to solve this.

dacodekid, 2019-02-18:
@atanisoft . That's the issue (I've enabled formatting which rearranged `includes`.  At one point I thought about that as well, but overlooked it. Thank you 👍.

closed 2019-02-18 by dacodekid

warlord0, 2019-03-27:
> There is a missing class type declaration in AsyncJson.h for sure, you will need to make sure you are including ESPAsyncWebServer.h _BEFORE_ AsynJson.h to solve this.

Many thanks, I've encountered this exact issue in Arduino IDE 1.8.9 and resolved it just by ordering my includes

```cpp
#include <ESPAsyncWebServer.h>
#include <AsyncJson.h>
#include <ArduinoJson.h>
```