me-no-dev/ESPAsyncWebServer #476: Webserver stops working when multiple URL's are requested.

TheNetStriker, 2019-02-18:
I'am encountering a problem on my Adafruit HUZZAH32 when providing a Webserver with this library and also with the non async version. I'am using the webserver to send 433 MHZ signals using the rc-switch library. This causes a few 100 miliseconds delay until the webserver can send the response.

If I only call one url at a time there is no problem with this. But when I try to send multiple request in a short time the webserver seems to crash. I'm using Apache JMeter for this and I can send about 10 requests to cause such a crash. Sadly I don't get any errors in the serial port console, the webserver just stops answering after those 10 requests.

I can also cause this crash if I just replace the line to send the 433 MHZ signal with a simple delay. The only way to get my code to work that I found so far is to create a QueueList. In this case I only add one element to the queue in webserver method which reduces the delay to respond. After that I proccess one element from the quere every time the loop method is called. This way I was able to send multiple http requests and send the messages without causing this crash.

Is this a known limitation of the webserver library or is there maybe a better solution for this?

hostmit, 2019-02-21:
Welcome to the club. Repo owner refuses to notify ESP32 users regarding crashes spontaneously when multiple request received. 
@TheNetStriker known issue. No fix available as far as I know.

AlessandroAU, 2019-03-02:
I Also appear to be having the same issue

wkraft-fablabka, 2019-03-04:
Showing code is always helpful, but you are giving some answer already yourself:
> I can also cause this crash if I just replace the line to send the 433 MHZ signal with a simple delay.

[Read the documentation:](https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember)
-  You can not use yield or delay or any function that uses them inside the callbacks

So -without code- the best guess is, that the rc-library makes use of delay as well.

TheNetStriker, 2019-03-04:
Thanks for the hint. I already guessed that the delay caused this problem, so I implemented a QueueList that will be filled in the async callback and the rc-library processes this list later.  But I also got some problems with that when sending a lot of requests to the webserver.

I now switched from a webserver to a simple MQTT client library, which is better suited for my use case. With this library I can also send events back to connected clients.

closed 2019-03-04 by TheNetStriker