me-no-dev/ESPAsyncWebServer #477: 'class WiFiClass' has no member named 'hostname

GeorgeFlorian, 2019-02-19:
Hello ! I just copy/paste your simple_server example and it doesn't work.

This is what I get from compiling.

```
Arduino: 1.8.8 (Linux), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

/home/george/New Volume/Arduino Sketches/foo2/foo2.ino: In function 'void setup()':
foo2:37:25: error: 'class WiFiClass' has no member named 'hostname'
     Serial.println(WiFi.hostname());
                         ^
Multiple libraries were found for "WiFi.h"
 Used: /home/george/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /opt/arduino-1.8.8/libraries/WiFi
exit status 1
'class WiFiClass' has no member named 'hostname'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

This is the code:

```
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

void setup() {

    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    Serial.print("Hostname: ");
    Serial.println(WiFi.hostname());

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world");
    });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [] (AsyncWebServerRequest *request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
    });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest *request){
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        } else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
    });

    server.onNotFound(notFound);

    server.begin();
}
void loop() {}
```

Pablo2048, 2019-02-19:
Did you notice this message from compiler?

> Multiple libraries were found for "WiFi.h"

It seems like you have installed library, which collide with ESP8266 core.

GeorgeFlorian, 2019-02-19:
> Did you notice this message from compiler?
> 
> > Multiple libraries were found for "WiFi.h"
> 
> It seems like you have installed library, which collide with ESP8266 core.

Of course I saw that. But it works with other WiFi sketches. I also did a quick search for the hostname function in both libraries and did not find it.
I removed the WiFi library from `/opt/arduino-1.8.8/libraries/WiFi` and now it should only use this one `/Arduino/hardware/espressif/esp32/libraries/WiFi`

```
Arduino: 1.8.8 (Linux), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

/home/george/New Volume/Arduino Sketches/foo2/foo2.ino: In function 'void setup()':
foo2:37:25: error: 'class WiFiClass' has no member named 'hostname'
     Serial.println(WiFi.hostname());
                         ^
exit status 1
'class WiFiClass' has no member named 'hostname'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Why do you say ESP8266 core if I'm using ESP32 ? Asking this because I am a complete noob.

atanisoft, 2019-02-19:
as mentioned on the other issue you opened https://github.com/espressif/arduino-esp32/issues/2479, the example should be updated to use the correct naming for the esp32 as it is not the same as the esp8266 WiFi library.

GeorgeFlorian, 2019-02-19:
> as mentioned on the other issue you opened [espressif/arduino-esp32#2479](https://github.com/espressif/arduino-esp32/issues/2479), the example should be updated to use the correct naming for the esp32 as it is not the same as the esp8266 WiFi library.

Indeed.

Changing from `Serial.println(WiFi.hostname());` to `Serial.println(WiFi.getHostname());` fixed the problem.
The example should be updated.

closed 2019-02-20 by GeorgeFlorian