me-no-dev/ESPAsyncWebServer #480: GET, POST and FILE parameters issue

GeorgeFlorian, 2019-02-20:
Hello ! First of all thank you for this library and your work.

I've been trying to save some values from two input fields inside a HTML page.

Here is main .cpp: gist.github.com/GeorgeFlorian/7ce3de245e8a39434a5effc92d8f53a8 
Here is the index.html: gist.github.com/GeorgeFlorian/d52ed6fb10e4beaf8f64ef5edef2a78b 

As you can see I've been trying to save those params in other `const *char` just to be able to `Serial.print();` them but it doesn't work.

Also, the `request->redirect("/url");` does not work as intended. I've pressed on the two buttons and on enter and nothing happens. 

Does anybody have any idea ? Because I really don't.

Thank you !

GeorgeFlorian, 2019-02-26:
Is my issue so hard to fix ? Did I break it ? I still have no idea how to make it work...

ruchuf, 2019-02-27:
First: reduce your code to the bare minimum needed, to show your problem.
Second: make yourself more familiar with C pointers and char arrays

The error message, you are showing, tells you, that there is no (longer a) connection to the ESP. That is the result of a exception, you should also see on the serial monitor. 
The cause of the exception is, that you try to print a pointer to a pointer to a char array.

Another tip: [Serving static files from SPIFFS](https://github.com/me-no-dev/ESPAsyncWebServer#serving-files-in-directory) helps you, reducing the code.

closed 2019-02-28 by GeorgeFlorian