me-no-dev/ESPAsyncWebServer #484: Corrupt Heap - Saving 2 string inputs from a HTML Form

GeorgeFlorian, 2019-02-28:
### Hardware:

Board: ESP32 DEVKIT1
Core Installation version: 1.0.1-git [this one](https://github.com/espressif/arduino-esp32)
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Linux Mint 19.1 Mate

Hello !
I am trying to save some values from two HTML inputs. But I get `CORRUPT HEAP` error and the ESP restarts. The strange thing is that it saves the values, but it still restarts, therefore making it unreliable.
Entering other values after the restart and hitting submit doesn't restart the ESP anymore. It restarts only on the first try.

Here is the full code:
```
#include <Arduino.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
 
const char* ssid = "ssid";
const char* password = "password";
String v[2];
 
AsyncWebServer server(80);
 
void setup() {
  Serial.begin(115200);
  delay(2000);
  if(!SPIFFS.begin(true)){
    Serial.println("Couldn't mount SPIFFS. Formatting... ");
  }  
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println(WiFi.localIP());

  server.on("/url", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index2.html", "text/html");
  });

  server.on("/login", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
  });

  server.on(
    "/login",
    HTTP_POST,
    [](AsyncWebServerRequest * request){
      int params = request->params();
      for(int i=0;i<params;i++){
        AsyncWebParameter* p = request->getParam(i);
        if(p->isFile()){ //p->isPost() is also true
          Serial.printf("FILE[%s]: %s, size: %u\n", p->name().c_str(), p->value().c_str(), p->size());
        } else if(p->isPost()){
            Serial.printf("POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
            request->redirect("/url");
            v[i] = p->value().c_str();
          } else {
              Serial.printf("GET[%s]: %s\n", p->name().c_str(), p->value().c_str());
            }
      }
      Serial.println("Printing v[i]: ");
      for(int i = 0; i<2; i++) {
        Serial.println(v[i]);
      }
      Serial.println("Printing ASCII: \n");
      String foo = v[0];
      for(int i=0; i<foo.length(); i++) {
        Serial.println((String)foo[i] + " " + int(foo[i]));
        }
      Serial.println();
 
      request->send(200);
    });
 
  server.begin();
}
 
void loop() {}
```
This is the [HTML](https://gist.github.com/GeorgeFlorian/d52ed6fb10e4beaf8f64ef5edef2a78b).

It seems that it crashes right after this part:

```
Serial.println("Printing v[i]: ");
for(int i = 0; i<2; i++) {
  Serial.println(v[i]);
  }
Serial.println("Printing ASCII: \n");
String foo = v[0];
for(int i=0; i<foo.length(); i++) {
Serial.println((String)foo[i] + " " + int(foo[i]));
}
```

This is the output:

```
Connecting to WiFi..
192.168.1.110
POST[networkName]: aqua2
POST[networkPassword]: 123456789
Printing v[i]: 
aqua2
123456789
Printing ASCII: 

a 97
q 113
u 117
a 97
2 50

CORRUPT HEAP: Bad head at 0x3ffcb340. Expected 0xabba1234 got 0x3ffcb57c
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400dc09b on core 0

Backtrace: 0x4008c990:0x3ffb4fa0 0x4008cbc1:0x3ffb4fc0 0x400dc09b:0x3ffb4fe0 0x40091f31:0x3ffb5010 0x4008458e:0x3ffb5030 0x400866f5:0x3ffb5050 0x4000bec7:0x3ffb5070 0x400fadfb:0x3ffb5090 0x400fae5b:0x3ffb50b0 0x400faea3:0x3ffb50d0 0x400fd596:0x3ffb50f0 0x400fd5fe:0x3ffb5110 0x4012b3cd:0x3ffb5130 0x400fa35c:0x3ffb5150 0x4008ec35:0x3ffb5180
```

Here is the stack decoded:
```
0x4008c990: commonErrorHandler_dump at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 495
0x4008cbc1: panicHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 235
0x400dc09b: __register_frame_info_table_bases at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 141
0x40091f31: multi_heap_realloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 240
0x4008458e: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 269
0x400866f5: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400fadfb: lwip_socket_register_membership at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3235
0x400fae5b: lwip_socket_unregister_membership at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3255
0x400faea3: lwip_socket_unregister_membership at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3256
0x400fd596: lwip_getsockopt_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3515
0x400fd5fe: lwip_close_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/sockets.c line 3537
0x400fa35c: create_msg at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 432
0x4008ec35: pxPortInitialiseStack at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 182

```

ruchuf, 2019-03-04:
Under certain conditions you are closing the request two times! You need to clean your code to make sure, that a request gets only one answer. Read [the documentation](https://github.com/me-no-dev/ESPAsyncWebServer#request-life-cycle) :
> When the Response is sent, the client is closed and freed from the memory

request->send() works on undefined (because of freed) memory, after request->redirect() has fired before!

gibo77, 2019-04-18:
I am just wondering if the use of String would contribute to crash. Read this thread if it is applicable. Correct me if I am wrong. 
http://forum.arduino.cc/index.php?topic=382944.msg2640860#msg2640860

ruchuf, 2019-04-18:
> I am just wondering if the use of String would contribute to crash.
Just a plain NO!
His application crashes, because of programming mistakes. In case of a POST request he finishes the request with an redirect:  `request->redirect("/url");` and tries to finish the exactly same request a second time with `request->send(200);`  But at this time, the request object is already gone and invalid!

The linked discussion is valid for a small Atmega based Arduino, where Flash and RAM are really limited!
The top argument, that you need 1600 bytes more flash is not valid here:  Because this library and for sure other ESP libraries are making heavy use of "string"-objects, therefore the string library is already linked to the program.  
The problem of heap fragmentation has nothing to do with strings at all, as strings can be used like char arrays on the stack and on the heap!  Its a problem in general, as that can be caused by any kind of data object, which gets dynamically allocated.  Here you need no nightmare-reports, better you investigate in designuing better algorithms and data structures.
E.g.   Any kind of parser should allocate one big chunk of memory, and implementing a own memory managemnt within this chung for parsing XML, JSON, HTMl, ....  and later free the big chunk.
And yes: each string operation like adding/removing characters ends up in additional malloc/free calls.

To learn about what is going on, monitor heap usage of your application, before panic!
**ESP.getFreeHeap(),  ESP.getHeapFragmentation()** and **ESP.getMaxFreeBlockSize()** are your friends!

GeorgeFlorian, 2019-04-18:
I did end up making it work:

```
      server.on("/dhcpLogin", HTTP_POST, [](AsyncWebServerRequest * request){
        
        int params = request->params();
        String y[2];
        for(int i=0;i<params;i++){
          AsyncWebParameter* p = request->getParam(i);
          if(p->isPost()){
              logOutput((String)"POST[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");            
              y[i] = p->value();            
            } else {
                logOutput((String)"GET[" + p->name().c_str() + "]: " + p->value().c_str() + "\n");
              }
        } // for(int i=0;i<params;i++)
        logOutput("Printing y[i] from dhcpLogin POST: ");
        for(int i = 0; i<2; i++) {
          logOutput(y[i]);
          }
        
        if(y[0] != NULL && y[0].length() != 0 &&
          y[1] != NULL && y[1].length() != 0) {
          File inputsWrite = SPIFFS.open("/inputs.txt", FILE_WRITE);
          if(!inputsWrite) logOutput((String)"ERROR_INSIDE_POST ! Couldn't open file to write DHCP IP credentials !");
          inputsWrite.println(y[0]);  // SSID
          inputsWrite.println(y[1]);  // Password
          inputsWrite.close();
          logOutput("Restarting in 5 seconds...");
          request->redirect("/logs");
          digitalWrite(LED, LOW);
          display.clearDisplay();
          display.print("Restarting in 5 seconds !");
          delay(5000);
          ESP.restart();
        } else request->redirect("/dhcpLogin");
      
      });
```

gibo77, 2019-04-19:
@ruchuf good thing to know that String() data types limitations are a thing of the past, due to the advent of larger RAM memory sizes such as ESP chips. I believe it is 1M byte?

Arduino.cc should start updating their web servers that is applicable to what is available or popular in the market.

ruchuf, 2019-04-23:
> good thing to know that String() data types limitations are a thing of the past, due to the advent of larger RAM memory sizes such as ESP chips. I believe it is 1M byte?

Don't muddle Flash with RAM!

The most common ESP8266 boards have 4MB of flash (between 512kB and 16MB available),  but only 80kB RAM, from which around 50kB are available fro user programs.

As String() objects are contained in this library, you have no penalty of using String() objects, as the corresponding library is already linked to the binary.  However you can still run into the problem, that heavy use of certain String()-methods leads to memory fragmentation.

closed 2019-10-05 by stale[bot]