me-no-dev/ESPAsyncWebServer #487: Support ArduinoJSON v6

Aircoookie, 2019-03-03:
Hello,

thanks again for the great async library!
Right now I'm adding JSON related communication to my application using the AsyncWebServer.
I've noticed that the examples use the ArduinoJSON v5 syntax and trying to use ArduinoJSON v6.9.1 with AsyncJson.h results in compilation errors:

```
In file included from C:\Users\cschwinne\git\Arduino\WLED02\wled00\wled00.ino:63:0:

C:\Users\cschwinne\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/AsyncJson.h:70:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     DynamicJsonBuffer _jsonBuffer;

           ^

In file included from C:\Users\cschwinne\git\Arduino\WLED02\wled00\wled00.ino:63:0:

C:\Users\cschwinne\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/AsyncJson.h:136:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

         DynamicJsonBuffer jsonBuffer;

```

On a related note, using the standalone `ArduinoJson-v5.13.5.h` (included after AsyncJson.h) did not work, because the AsyncJson.h includes <ArduinoJson.h> itself. Guess there is not much one can do about that, so actually installing the lib seems to be required.

Since ArduinoJSON v6 is out of beta now, it would be great to have it supported soon. If that isn't possible, it would be nice to add a note to the readme that v5 is explicitely required.

By the way, I did not yet do more than minimal testing, but it seems like v5.13.5 is working as intended.

Thank you!

wkraft-fablabka, 2019-03-04:
I recently investigated a little bit in that problem. The biggest difference in ArduinoJSON from V5 to V6 is, that the dynamic memory allocation from JSONbuffer is now missing.
So the constructors of the JSON-related classes need
- either a fixed size for the new JsonDocument
- or better an additional parameter, how big the buffer should be
A good point would be, to make use of the #define ARDUINOJSON_VERSION respective ARDUINOJSON_VERSION_MAJOR, so V5 and V6 could be used in parallel.

Aircoookie, 2019-03-04:
Thanks for your quick reply!
That's good, it would definitely be possible to use that header to distinguish between the version the user has installed. The fact that the buffer is fixed in size in V6 is not a bad thing though, since it mitigates potential heap fragmentation issues. I agree with you that it would be a good idea to have a parameter for selecting the buffer size, since user requirements will vary wildly. Probably the easiest to use and implement would be an optional constructor parameter like:

```cpp
AsyncJsonResponse * response = new AsyncJsonResponse(1024);
```

which then internally creates a `DynamicJsonDocument doc(1024);`

baggior, 2019-03-12:
ArduinoJson 6 is now released (not in beta)
and my project that depends on this have to be migrated due to  library automatic update

I have a problem with ESP Async WebServer library and particularly with AsyncJson.h file that is not compatible with ArduinoJson 6 

are you intended to migrate soon?

baggior, 2019-03-17:
I've created a pull request to fix it #491 

please merge

Aircoookie, 2019-03-17:
Thanks for creating that compatibility PR! I'm not a contributor to EspAsyncWebServer, still I would propose that we make `DYNAMYC_JSON_DOCUMENT_SIZE` an (optional) parameter for `AsyncJsonResponse` and `AsyncCallbackJsonWebHandler`, since hard-coding too much heap allocation might cause issues in complex apps and too little will cause many applications to not be able to (de)serialize. 1024 sounds like a good baseline for most simple stuff, but having that configurable would be a good idea.
Also, we can make use of `#define ARDUINOJSON_VERSION` to switch implementation automatically based on whether the user has ArduinoJSON v5 or v6.

I might do another PR based on yours later once I have time :)

baggior, 2019-03-28:
@Aircoookie, I think I've done what you proposed on commit ccc57034365dc470133f691e74fe5c23c437758c

baggior, 2019-03-29:
Also added compatibility macro for old ArduinoJson v5 
#491

Aircoookie, 2019-03-29:
Thank you very much, looks really promising! Will give it a shot soon!
Just one small suggestion (I don't have write access to your PR): Replace all `#ifdef ARDUINOJSON_5_COMPATIBILITY` with `#if ARDUINOJSON_VERSION_MAJOR == 5`. That way, users don't have to add the define themselves, it's fully automatic.

baggior, 2019-04-01:
@Aircoookie OK
I did what you suggested

closed 2019-10-05 by stale[bot]