me-no-dev/ESPAsyncWebServer #490: WebSocket setAuthentication doesn't work

Kolsha, 2019-03-15:
Hi. I faced a bug that websocket authorization does not work in mobile Google chrome app (72.0.3626.101 version). I get "Disconnect" when I use setAuthentication. In the desktop version of Google chrome all works good.

Kolsha, 2019-07-26:
Anyone?

polaco1782, 2019-08-04:
> Anyone?

Yes, Im having the same issue. Seems the problem is on the username/password validation.

on the file AsyncWebSocket.cpp, line 1157, the code:

`if((_username != "" && _password != "") && !request->authenticate(_username.c_str(), _password.c_str()))`

seems to be the problem. If you comment out the (_username != "" && _password != "") it will ask for the credentials. I've not tested it enough, I'll try to fix it and write a patch

closed 2019-10-17 by stale[bot]

Kolsha, 2020-02-03:
@me-no-dev Is this fixed?