me-no-dev/ESPAsyncWebServer #495: Esp crash looping when sending event while testing SSEs

Shadow5112, 2019-03-23:
I'm trying to get the provided SSE example working but the ESP keeps crash looping every time that it tries to send an event.  

source.cpp
```
void initServer(AsyncWebServer* server, ParamsStruct* params) {
    //Create Access Point
    WiFi.softAP("MyESP32AP", "testpassword");
    Serial.println();
    Serial.print("IP address: ");
    Serial.println(WiFi.softAPIP());

    
    AsyncEventSource events("/events");
    DefaultHeaders::Instance().addHeader("Access-Control-Allow-Origin", "*");

    /* XHR Example.
        - Param "name" is sent from mission control.
        - "name" is copied to the params object
        - params object is then passed to vSayHelloTask - see main.cpp
        - vSayHello task then accesses name directly.

        Note: for ANY parameters you want to use, you must add them to
        the paramsStruct struct located in Source.h first. 
    */
    server->on("/update_name", HTTP_POST, [=](AsyncWebServerRequest *request){
        strcpy(params->name, request->arg("name").c_str());
        request->send(200, "text/plain", "Success");
    });
    
    
    /* SSE Example.
        - SSEs will be used to continuously send data that was
        not necessarily requested by mission control
        (e.g. temperature, something we should send periodically)

        - Once mission control declares the ESPs IP address at a certain 
        endpoint to be an EventSource, the ESP can trigger events on the web
        interface, which the web interface can attach event listeners to
        (similar to how we are attaching event listeners for when we recieve
        XHRs to /update_name above, allowing us to do things when we recieve an 
        XHR).
        - Below's example is an example of sending SSEs when mission control
        declares our ip address and endpoint (e.g. 192.168.4.1/events) to be
        an event source.
        - More info on this concept here: 
            https://developer.mozilla.org/en-US/docs/Web/API/EventSource
    */
    
    events.onConnect([](AsyncEventSourceClient *client) {
      if(client->lastId())
      {
        Serial.printf("Client reconnected! Last message ID that it gat is: %u\n", client->lastId());
      }
      // send event with message "hello!", id current millis
      // and set reconnect delay to 1 second
      client->send("hello!", NULL, millis(), 1000);
      delay(1000);
      client->send("hello!", NULL, millis(), 1000);
      delay(1000);
      client->send("hello!", NULL, millis(), 1000);
      delay(1000);
      client->send("hello!", NULL, millis(), 1000);
      delay(1000);
    });
 
    //Attach event source to the server.
    server->addHandler(&events);

    //Start server.
    server->begin();
}

```

This bit of code was cut and pasted directly from the GitHub example.

index.html
```
<!DOCTYPE html>
<html>
<head>
	<title>SSE Test</title>
	<script type="text/javascript">
		if (!!window.EventSource) 
	{
	  	var source = new EventSource('http://192.168.4.1/events');

		source.addEventListener('open', function(e) {
		    console.log("Events Connected");
			  }, false);

		source.addEventListener('error', function(e) {
		    if (e.target.readyState != EventSource.OPEN) {
		      console.log("Events Disconnected");
		  	  }
		  }, false);

		  source.addEventListener('message', function(e) {
		    console.log("message", e.data);
		  }, false);

		  source.addEventListener('myevent', function(e) {
		    console.log("myevent", e.data);
		  }, false);
	}
	</script>
</head>
<body>

</body>
</html>
```

This is the error message that the ESP puts out to the monitor. It only starts doing this after it connects with the client. 
```
IP address: 192.168.4.1
Successfully initialized EEPROM, size = 100.
I (630) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (640) wifi: station: f8:28:19:d5:29:a3 join, AID=1, bgn, 40U
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060b30  A0      : 0x8010232e  A1      : 0x3ffd0560  
A2      : 0x3ffb4010  A3      : 0x3ffc0fd0  A4      : 0x3ffce6c8  A5      : 0x3ffc0898  
A6      : 0x00ff0000  A7      : 0x00000000  A8      : 0x800ff2c8  A9      : 0x3ffd0520  
A10     : 0x3ffc14dc  A11     : 0x3ffc0fd0  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffd0560 0x4010232b:0x3ffd05a0 0x4010244f:0x3ffd05e0 0x401026b1:0x3ffd0640 0x40160a09:0x3ffd0660 0x40160a9d:0x3ffd0690 0x40160af1:0x3ffd06b0 0x40160c3e:0x3ffd06d0 0x40160cb0:0x3ffd06f0 0x4008c485:0x3ffd0720
0x4010232b: AsyncWebServerRequest::_parseLine() at /home/livi/controlsystems2019/Projects/SSE/components/ESPAsyncWebServer/src/WebRequest.cpp:990

0x4010244f: AsyncWebServerRequest::_onData(void*, unsigned int) at /home/livi/controlsystems2019/Projects/SSE/components/ESPAsyncWebServer/src/WebRequest.cpp:990

0x401026b1: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#7}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /home/livi/controlsystems2019/Projects/SSE/components/ESPAsyncWebServer/src/WebRequest.cpp:990
 (inlined by) _M_invoke at /home/livi/controlsystems2019/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1871

0x40160a09: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /home/livi/controlsystems2019/Projects/SSE/components/AsyncTCP/src/AsyncTCP.cpp:870

0x40160a9d: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /home/livi/controlsystems2019/Projects/SSE/components/AsyncTCP/src/AsyncTCP.cpp:870

0x40160af1: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /home/livi/controlsystems2019/Projects/SSE/components/AsyncTCP/src/AsyncTCP.cpp:870

0x40160c3e: _handle_async_event(lwip_event_packet_t*) at /home/livi/controlsystems2019/Projects/SSE/components/AsyncTCP/src/AsyncTCP.cpp:870

0x40160cb0: _async_service_task(void*) at /home/livi/controlsystems2019/Projects/SSE/components/AsyncTCP/src/AsyncTCP.cpp:870

0x4008c485: vPortTaskWrapper at /home/livi/controlsystems2019/esp-idf/components/freertos/port.c:403

Rebooting...
```

This is the error messages that get thrown on the client side.

```
GET http://192.168.4.1/events net::ERR_CONNECTION_RESET
index.html:16 Events Disconnected
192.168.4.1/events:1 
GET http://192.168.4.1/events net::ERR_CONNECTION_RESET
index.html:16 Events Disconnected
192.168.4.1/events:1 
GET http://192.168.4.1/events net::ERR_CONNECTION_RESET
index.html:16 Events Disconnected
192.168.4.1/events:1 
GET http://192.168.4.1/events net::ERR_CONNECTION_RESET
index.html:16 Events Disconnected
192.168.4.1/events:1 

```

zhivko, 2019-05-03:
Have you tried with esp-idf release/3.2

philbowles, 2019-05-30:
I dont think you can call delay() inside an asynchronous callback. In fact I know you can't on ESP8266, When you think bout it, it defeats the whole point of async programming in the first place

Pablo2048, 2019-05-30:
Why people aren't reading documentation first?
https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember

me-no-dev, 2019-06-23:
biggest issue is that you have `AsyncEventSource events("/events");` inside the init function, which means that as soon as you exit that function, the event source will be gone.

closed 2019-06-23 by me-no-dev