me-no-dev/ESPAsyncWebServer #497: Question:  How to make an OTA firmware update , using your ASYN libraries ( not via arduino IDE) ?

gnkarn, 2019-03-28:
i was looking for an example on how to insert firmware OTA update , using your async version and a browser ( not using arduino IDE), but couldn't find an example , all i found were with the non async libraries or using the arduino IDE .

Appreciate , if you have a link of an example at hand .

thanks

lbernstone, 2019-03-31:
Written for ESP32, but should work on ESP8266 (may need minor change to begin command).  https://github.com/lbernstone/asyncUpdate

dushyantahuja, 2019-09-27:
+lbernstone - love your functions - unfortunately, I'm running into a small issue - can you please check. 

https://github.com/lbernstone/asyncUpdate/issues/3

> Written for ESP32, but should work on ESP8266 (may need minor change to begin command). https://github.com/lbernstone/asyncUpdate

closed 2019-12-10 by stale[bot]

chess-levin, 2023-02-10:
Could anyone provide a working example how to integrate OTA with an Upload HTML-form on an AsyncWebServer ?

HHavendano, 2023-03-23:
> Escrito para ESP32, pero debería funcionar en ESP8266 (puede necesitar un cambio menor para comenzar el comando). https://github.com/lbernstone/asyncUpdate

Hi, i tried the code ,and it's working. but i wonder if i can change the HTML ("void handleUpdate"). i don´t wanna create the HTML on the code. i wanna add an existing website where it will work the same.