me-no-dev/ESPAsyncWebServer #503: Error 500 send files from spiffs

steinwedel, 2019-04-18:
I am using the following snippet, to send files from SPIFFS to the web browser:

`
AsyncWebServerResponse *response = request->beginResponse(*_fs, path, contentType);
request->send(response);
`

The web page calls several js and css from the web server. The web server returns for some of the files error 500. It is random which request ends with error 500. A reload changes the result.
After moving files to PROGMEN, problem is gone. Obviously it has to do with the number of open files. I thing the server is not checking, how many files are open ESP8266 has limit of 5 and  returns a 500 if opening fails.
Is there a way to delay a request if a max number of open files is reached.
a code snipped would be fine.

Probably related: #412 ,#349

PotatoSauceVFX, 2019-04-20:
Same issue here. :( I don't get a 500 error though, the web server just stops responding completely until I reset.

closed 2019-10-05 by stale[bot]

1024kilobyte, 2020-03-04:
The issue still seems to exist. Just ran into it (latest version to date). So its not possible to serve the website as random files are missing...