me-no-dev/ESPAsyncWebServer #504: Too many messages on websocket connection

iotlearner0level, 2019-04-24:
Sorry if this question has been asked before, i could not find a solution. here is my program ( an ina219, with its data displayed in the browser using websockets) https://github.com/iotlearner0level/INA219-sensor

but i can't achieve more than ~15 messages/second. I want to pass more than 50 readings (V,I, Power) per second for realtime display.

This is what i get

```
:43.651 -> ERROR: Too many messages queued
16:47:43.752 -> ERROR: Too many messages queued
16:47:43.823 -> ERROR: Too many messages queued
16:47:43.897 -> ERROR: Too many messages queued
16:47:43.974 -> ERROR: Too many messages queued
16:47:44.856 -> ERROR: Too many messages queued
16:47:44.956 -> ERROR: Too many messages queued
16:47:45.026 -> ERROR: Too many messages queued
16:47:45.095 -> ERROR: Too many messages queued
16:47:45.163 -> ERROR: Too many messages queued
16:47:45.230 -> ERROR: Too many messages queued
16:47:45.300 -> ERROR: Too many messages queued
16:47:45.402 -> ERROR: Too many messages queued
16:47:45.469 -> ERROR: Too many messages queued
16:47:45.550 -> ERROR: Too many messages queued
16:47:45.623 -> ERROR: Too many messages queued
```
if i decrease the send interval.

i may be missing something, but please not that i am another hobbyist and not a professional programmer. i don't have a thorough understanding of networking/communications/programming/memory and all that stuff.

philbowles, 2019-05-30:
I'm guessing you are on ESP32, ESP8266 has hard-coded limit of 8!

iotlearner0level, 2019-05-30:
ok, now i get it. i am using esp8266 (wemos d1 mini clone) and i get maximum of about 15 messages/sec. my goal was to see if i could plot realtime graph as a replacement of lcd like cnlohr did with the esp api https://www.youtube.com/watch?v=8ISbmQTbjDI

i am not sure but maybe if it would be possible to send multiple readings per message to get more than 15 messages maybe?

philbowles, 2019-05-30:
It may be, but then you run into another issue where the volume of data being sent can rapidly reduce the free heap...you will have to experiment to find a good balance.

philbowles, 2019-05-30:
so you will have to be quite loose with what you mean by  realtime . The other option is to use SSE . that code is not "crippled" yet and will go a lot further before getting overloaded. See here for more info. https://github.com/me-no-dev/ESPAsyncWebServer/pull/509

philbowles, 2019-05-30:
Watch this at about 8:30 to see SSE in action https://www.youtube.com/watch?v=Wv3i_y-9Q-c

me-no-dev, 2019-06-23:
given that you as human will not be able to process 50 messages a second, why not buffer them a bit and send every 100ms or so?

moothyknight, 2019-09-05:
I'm getting this same issue running running the websocket at about 13 messages/sec outputting strings. 

```
void wsTask(void *p){
  while(globalClient != NULL && globalClient->status() == WS_CONNECTED){
    globalClient->text(output);
    vTaskDelay(75 / portTICK_PERIOD_MS);
  }
  vTaskDelete(NULL);
}
```
It's sending strings like this: 1253928|1187.00|1329.00|0.8932|0.9202|0.8863|1189|-0.0007|0.0000|55.0000
which include time, sensor readings, and some math.

The delay works just fine to run the websocket asynchronously, however it crashes if the connected client lags in processing the websocket queue. Is there a way to clear the buffer every second? Or would it be better practice to wait for a feedback message from the connected client to add to the queue?

atanisoft, 2019-09-06:
@moothyknight it may be better to use an EventSource instead of a WS.

moothyknight, 2019-09-06:
@atanisoft 

Thanks for plugging that hole in my knowledge, the EventSource example in the docs worked like a charm! I'm getting 50ms updates on my sensor now which is all I need, haven't pushed it higher since I don't need to, what is the maximum you think?

atanisoft, 2019-09-06:
The upper limit would be determined by the speed at which the updates can be broadcast to the connected consumers. If you are hitting ~50ms between updates I'd say that is probably pretty good.

closed 2019-11-19 by stale[bot]