me-no-dev/ESPAsyncWebServer #505: Impossible to build with ESP32 1.0.1 and AsyncTCP 1.0.3

christophepersoz, 2019-04-26:
Hello,

I'm currently facing to this issue when trying to bulked a simple WebSocket example on my ESP WEMOS LOLIN 32.

I had a look on several issues reported, all of them said to compile AsyncTCP 1.0.3 with ESP32 1.0.1, but is seems that there is a another problem though.
Any idea to make it works?

```
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:115:8: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   addr.addr = ip;
        ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:148:35: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     return connect(IPAddress(addr.addr), port);
                                   ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'bool AsyncClient::operator==(const AsyncClient&)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:192:66: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) && (_pcb->remote_port == other._pcb->remote_port));
                                                                  ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:192:96: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) && (_pcb->remote_port == other._pcb->remote_port));
                                                                                                ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCPbuffer.cpp: In constructor 'AsyncTCPbuffer::AsyncTCPbuffer(AsyncClient*)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCPbuffer.cpp:35:15: error: 'panic' was not declared in this scope
         panic();

               ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'void AsyncClient::_dns_found(const ip_addr*)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:473:31: error: 'const struct ip_addr' has no member named 'addr'
     connect(IPAddress(ipaddr->addr), _connect_port);
                               ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getRemoteAddress()':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:586:26: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return _pcb->remote_ip.addr;
                          ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'uint32_t AsyncClient::getLocalAddress()':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:598:25: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return _pcb->local_ip.addr;
                         ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCPbuffer.cpp: In member function 'virtual size_t AsyncTCPbuffer::write(const uint8_t*, size_t)':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCPbuffer.cpp:128:23: error: 'panic' was not declared in this scope
                 panic();

                       ^
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
/Users/cp/Developer/_Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp:873:14: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   local_addr.addr = (uint32_t) _addr;
              ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/cp/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
```

Thanks

Pablo2048, 2019-04-26:
Maybe I do have some idea. Did You read this compiler warning?
```
Multiple libraries were found for "WiFi.h"
 Used: /Users/cp/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
```
especially **Multiple libraries were found for "WiFi.h"**

Try to remove the one from .../Java/libraries/WiFi

christophepersoz, 2019-04-26:
Good idea, but that's not the problem, build error remains.
As it was mentioned ```Not used``` by GCC, I imagined that it was simply ignored.

The ```panic()``` issue can be easily removed by commenting the line, but it's not so simple for the ```addr.addr``` issue.

Thanks

Pablo2048, 2019-04-26:
It definitely IS the problem - I'm building with latest ESP32 Core, ESP Async WebServer 1.2.0 and AsyncTCP 1.0.3 without any issue. I think that the IDE messed up header files from both libraries on your computer.

christophepersoz, 2019-04-26:
Which version of Arduino IDE are you using? I'm currently on 1.8.9, ESP Async Webserver 1.2.0 and AsyncTCP 1.0.3.

christophepersoz, 2019-04-26:
Start digging and found this in ESP8266 libraries,

`ip_addr_t` is declared as struct under the lib `lwip/id_addr.h` of the Espressif 8266 package, 

```c++
/* This is the aligned version of ip_addr_t,
   used as local variable, on the stack, etc. */
struct ip_addr {
  u32_t addr;
};

/** ip_addr_t uses a struct for convenience only, so that the same defines can
 * operate both on ip_addr_t as well as on ip_addr_p_t. */
typedef struct ip_addr ip_addr_t;
typedef struct ip_addr_packed ip_addr_p_t;

```

Whereas the ESP32 libs is a bit different, I haven't found the same struct for` ip_addr_t`,
```c++
#if LWIP_IPV4 && LWIP_IPV6
/**
 * @ingroup ipaddr
 * A union struct for both IP version's addresses.
 * ATTENTION: watch out for its size when adding IPv6 address scope!
 */
typedef struct ip_addr {
  union {
    ip6_addr_t ip6;
    ip4_addr_t ip4;
  } u_addr;
  /** @ref lwip_ip_addr_type */
  u8_t type;
} ip_addr_t;
``` 

I'm not good enough to determine if that's the reason of the errors on build, anyway, there is a problem on this point especially.

Pablo2048, 2019-04-27:
I'm not using Arduino IDE (it is too simple for all of my needs). I'm using VSC+PIO with latest ESP32 Core. Please try to install portable version of Arduino IDE with only ESP32 Core library installed.

makeSmartio, 2019-06-11:
Try removing ESPAsyncTCP from you library. That is what was causing the problem for me

Theghost90, 2019-07-20:
Remove EAPAsyncTCP library from AsyncTCP package who is built for ESP32 MCU.
In AsyncTCP library (for ESP32) change ESPAsyncTCP to AsyncTCP

Theghost90, 2019-07-20:
Removing first the ESPAsyncTCP library helped me to perfectly troubleshoot and makes me understand any relationship between ESPAsyncTCP, AsyncTCP libraries and ESPAsyncWebServer library. Understand that the ESPAsyncWebServer library is build based on ESPAsyncTCP library if you use ESP8266 MCUs and based on AsyncTCP library if you use ESP32 MCUs.

So if you use the ESP32, No need to Remove the ESPAsyncTCP library from your Arduino library because ESPAsyncTCP library is built for ESP8266 MCUs. Thus in order to do not have any compilation errors, when you use ESP32 MCU simply make the following changes in the AsyncTCP library:

Rename ESPAsyncTCP.cpp and ESPAsyncTCP.h files to AsyncTCP.cpp
AsyncTCP.h files

Rename ESPAsyncTCPbuffer.cpp and ESPAsyncTCPbuffer.h files to AsyncTCPbuffer.cpp
AsyncTCPbuffer.h files

In AsyncPrinter.cpp and SyncClient.cpp files, change the library ESPAsyncTCP.h to AsyncTCP.h.

Once done, Inside every files correct ESPAsync to Async in order to get everything linked properly

Also commenting panic() method helped me.

The only update you have to make it's regarding Arduino OTA. If you upgrade to the Arduino ESP32 latest library, make sure to update the Arduino OTA also to the latest.

Thanks guys

closed 2019-10-05 by stale[bot]