me-no-dev/ESPAsyncWebServer #506: authentication

mybigman, 2019-04-27:
```cpp
// HTTP basic authentication
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    if(!request->authenticate(http_username, http_password))
        return request->requestAuthentication();
    request->send(200, "text/plain", "Login Success!");
});
```
Is there a way of authenticating globally or top level down "/" or do I need to call ```request->authenticate``` on every end point?

thanks.

GeorgeFlorian, 2019-05-09:
I would also like to know this.
I need to implement a simple username and password for my `AsyncWebServer server(80);`.

mybigman, 2019-09-23:
any ideas?

me-no-dev, 2019-09-23:
honestly, this is the only way currently. The other option is to create your own Handler class and do it there once (have a look at the AsyncFSEditor: https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L436 (if you need upload, that should have it also as uploads are done before the request hits the callback)

closed 2019-12-06 by stale[bot]