me-no-dev/ESPAsyncWebServer #510: Ignore response

renerlemes, 2019-05-09:
I have the followinn code:`

```
webServer.on("/reset", HTTP_POST, [=](AsyncWebServerRequest * request)
{
	AsyncWebServerResponse* response = request->beginResponse(200, "text/plain", "Device | Restart...");
	response->addHeader("Connection", "close");
	request->send(response);

       //Here call the function that restart ESP8266. But the ESP8266 restart is called, but the response not show, how to show response, then restart ESP8266?
});
```

How to response, then restart?

Pablo2048, 2019-05-09:
It's pretty simple. Just use Ticker library and use it for example like this:
```C
#include <Ticker.h>
...
Ticker deferred;
...
...
// Here call the function that restart ESP...
  deferred.once_ms(700, []()
  {
     ESP.restart();
  });
```

renerlemes, 2019-05-09:
@Pablo2048 the only way to do this is using Ticker library? Is possible to use a handler or callback to call when the response complete, then restart ESP. Or yet if the response object has a property that says you have finished?

Pablo2048, 2019-05-09:
Probably not - there can be more ways to do the same behavior if you put some time to study the source code, but I don't see any way to do this without library modification.

GeorgeFlorian, 2019-05-09:
In my project I've used a global `bool` flag that changes its value when I need a reset. I check for it in `void loop()`.
```
#include <Arduino.h>
bool restartFlag = false;
.
.
.
void setup () {
webServer.on("/reset", HTTP_POST, [=](AsyncWebServerRequest * request)
{
	AsyncWebServerResponse* response = request->beginResponse(200, "text/plain", "Device | Restart...");
	response->addHeader("Connection", "close");
        restartFlag = true;
	request->send(response);
});

}

void loop () {
if(restartFlag) {   // if(restartFlag == true)
ESP.restart();
}
```
}

Pablo2048, 2019-05-09:
Yes, but this method is way too fast for TCP stack to send the answer actually (not enough time for TCP ACK mechanism), so it probably wont work correctly...

GeorgeFlorian, 2019-05-09:
> Yes, but this method is way too fast for TCP stack to send the answer actually (not enough time for TCP ACK mechanism), so it probably wont work correctly...

I don't have any problems of yet using this method. If the page loads and the ESP restarts the page will remain.

renerlemes, 2019-05-09:
@Pablo2048 your code work. Tks. Abou the example of @GeorgeFlorian, the page not will remain and ESP restart before response show. After put a delay the code work.

The library could have a handler or callback for when the response is complete. This library implementation would be good.

Pablo2048, 2019-05-09:
You never notice any problem when you have so small payload as just this simple answer and header, and you rely on actual browser(s) behavior which is that also unfinished page transfer (without FIN/FIN ACK handshake) will be displayed. Adding delay solves this problem anyway...

hreintke, 2019-05-09:
You also can use `schedule_function` to execute code in the loop.

```
webServer.on("/reset", HTTP_POST, [=](AsyncWebServerRequest * request)
{
	AsyncWebServerResponse* response = request->beginResponse(200, "text/plain", "Device | Restart...");
	response->addHeader("Connection", "close");
        schedule_function([](){delay(2000);ESP.restart();}); 
	request->send(response);
});
```
The scheduled function will be executed after the current loop cycle.

GeorgeFlorian, 2019-05-09:
> You also can use `schedule_function` to execute code in the loop.
>
> ```
> webServer.on("/reset", HTTP_POST, [=](AsyncWebServerRequest * request)
> {
> 	AsyncWebServerResponse* response = [...]

I am certain that @me-no-dev specified to not use any `delays` inside those functions.

atanisoft, 2019-05-09:
> I am certain that @me-no-dev specified to not use any `delays` inside those functions.

The delay is not executed in this callback, it is scheduled and executed from the loopTask (which calls setup() and loop()).

renerlemes, 2019-05-09:
Using delays only, without schedule_function, the ESP return a exception and not working. Maybe @me-no-dev have a solution or suggestion for this issue. But, the best way until now, is the solution of @Pablo2048

hreintke, 2019-05-09:
@renerlemes 
Indeed, delay(..) directly within the callback gives exceptions. 
That is exactly the reason why I showed the use of schedule_function.

As @atanisoft mentioned, schedule_functions are not executed within the callback itself.
It it executed from the looptask, which in essence is 
```
while (true)
{
   loop();
   run_scheduled_functions()
}
```
There is no functional difference between 
- using the flag and test in loop
- using scheduled_function

I myself use the scheduled function to keep all my code at the same location and not split between callback and loop(). 
Just personal preference.

GeorgeFlorian, 2019-05-09:
> @renerlemes
> Indeed, delay(..) directly within the callback gives exceptions.
> That is exactly the reason why I showed the use of schedule_function.
>
> As @atanisoft mentioned, schedule_functions are [...]

Than that is a great solution. I had no knowledge of scheduled functions. I will try implement it in my projects as well.

renerlemes, 2019-05-09:
I´ll test too and maybe implement in my project. As you say @atanisoft, ..."keep all my code at the same location..."
Tks

atanisoft, 2019-05-09:
it should be noted that schedule_function is specific to the esp8266 at this time. The scheduler is not currently integrated into the esp32 stack.

me-no-dev, 2019-06-23:
so there is a way :) not easy though... one must implement their custom response or handler class. Then in the destructor of the class, call restart. That means that once the connection has been terminated, restart will be called.

closed 2019-10-05 by stale[bot]