me-no-dev/ESPAsyncWebServer #516: Server unreachable after ESP.restart();

GeorgeFlorian, 2019-05-21:
Hello !

After I perform an ESP.restart(); I can't connect to the WebServer. The page keeps loading for ever.
Do I need to close something before ESP.restart() ? Because this doesn't happen when I press on Reset Button.

GeorgeFlorian, 2019-05-28:
Hello ?

philbowles, 2019-06-14:
You need to power-cycle your device after a serial upload. If you don't ESP.restart will hang on next boot. I don't think it's a problem with this library.  See https://github.com/esp8266/Arduino/issues/1722

GeorgeFlorian, 2019-06-14:
> You need to power-cycle your device after a serial upload. If you don't ESP.restart will hang on next boot. I don't think it's a problem with this library. See [esp8266/Arduino#1722](https://github.com/esp8266/Arduino/issues/1722)

Who said anything about Serial Upload ?
I am talking about `ESP.restart();`. My program will sometimes make the ESP restart, and the WebServer hangs, it doesn't reset.

me-no-dev, 2019-06-23:
this makes no sense. there is nothing in the server to do it. When you restart all starts clean. You sure wifi and all is OK?

GeorgeFlorian, 2019-06-23:
> this makes no sense. there is nothing in the server to do it. When you restart all starts clean. You sure wifi and all is OK?

I did not check WiFi connection using WireShark or something similar. I will have to test more.

closed 2019-07-03 by GeorgeFlorian

aleuarore, 2021-05-12:
Hello, I'm having the same problem, did you fix it?

GeorgeFlorian, 2021-05-12:
> Hello, I'm having the same problem, did you fix it?

I honestly do not remember. What I do know is that it now works.
I have a flag that I switch to true whenever I need the ESP to restart.
I check for that flag inside `void loop()`.
Before `ESP.restart()` I also use `server.reset()`.
```
  if (shouldReboot) {
    Serial.println("Restarting in 5 seconds...");
    delay(5000);
    server.reset();
    ESP.restart();
  }
```

Hope this helps in some way.

aleuarore, 2021-05-12:
Super! It works.
Thanks a lot!!