me-no-dev/ESPAsyncWebServer #519: Issue with using EEPROM with Server.on Event Handler

MaxBayne, 2019-05-26:
i use AsyncWebServer and its working perfectly and i use on Event To Register handler like below

```
//GET: http://192.168.1.120/test
    server.on("/test", HTTP_GET, [](AsyncWebServerRequest *request) 
    {

      EEPROM.put(500,15);
      EEPROM.commit();

      
      request->send(200, "text/plain","Good Morning");
    });
```

when i use the previous code the ESP32 crashed with memory error i think handler need permission to access memory

and this is the error log

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4014d850: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalGuru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (Double exception)
```

any help please ?

lbernstone, 2019-05-27:
The first error indicates that you are trying to access flash during an interrupt, which is not allowed (it should raise a semaphore or flag and be processed outside the interrupt).  The double exception at the end means that something catastrophic happened such that the panic routine couldn't even run, which is why there is no backtrace info.  This can happen if you run out of stack memory.
What version of arduino-esp32 are you using?  1.0.3 (the git repo) has changed EEPROM to use nvs.

MaxBayne, 2019-05-27:
thanks my friend

my ESP32 Wroom32
and i use visual studio code with platformio 
i have arduino ide 1.8.9 but i dont use it just use visual code and worked with me fun
i tried to use [ESP32WebServer] and inside request handler On i call EEPROM Commands and its working perfectly but when i try to use [ESPAsyncWebServer] it fail and give me the error in main thread i think some thing worng inside ESPAsyncWebServer about give permission to access EEPROM inside Request Handler ...

gibo77, 2019-05-28:
Maybe use volatile in interrupt for the variable and exit right away.

MaxBayne, 2019-05-29:
how can i use it ?

philbowles, 2019-05-30:
Why on earth are you using EEPROM on an ESP32?

MaxBayne, 2019-05-31:
> Why on earth are you using EEPROM on an ESP32?

why not ?

philbowles, 2019-06-15:
> > Why on earth are you using EEPROM on an ESP32?
> 
> why not ?

a) Because it doesn't actually have any :)
b) it is emulated in flash for backwards compatibility with ancient code
c) SPIFFS moves writes around to level the wear, EEPROM library doesn’t.
d) It has problems like this

I hope that answers your questions, but anyway you should not be calling anything in an asynchronous callback that you wouldn't/couldn't call in an ISR. Set a global flag and deal with the event in the main loop, then reset the flag.

me-no-dev, 2019-06-23:
@philbowles EEPROM has been rewritten to use NVS on ESP32 and no longer has it's own partition. No longer suffers from wear leveling issues either :)

foxabilo, 2019-07-05:
Just a little advisory on this topic, not sure if it means anything, but the arduino ide (even the latest version 1.8.9) has the SPIFFS .CSV files specifying EEPROM locations to use even though if you look in the GIT the .CSV's have been updated to remove EEPROM from spiffs 🤷‍♂

closed 2019-10-05 by stale[bot]