me-no-dev/ESPAsyncWebServer #520: ESP32: FS.h: No such file or directory

andig, 2019-05-28:
I'm unable to compile on Platformio using git version of espressif32 platform:

    .piolibdeps/ESP Async WebServer_ID306/src/ESPAsyncWebServer.h:27:16: fatal error: FS.h: No such file or directory

I've already added:

````
lib_deps=
    ${common.lib_deps}
    https://github.com/me-no-dev/AsyncTCP
    SPIFFS
````

but it is not sufficient to fix the missing `FS.h`. Any ideas?

atanisoft, 2019-05-28:
You need to add FS as a dependency since PIO doesn't automatically include it as a dependency of SPIFFS.

andig, 2019-05-28:
Thanks, resolved. Or put `lib_ldf_mode=deep`.

closed 2019-05-28 by andig

Valentin-Metz, 2021-03-08:
I got the same error today installing via platformio. Maybe add it to the dependencies by default?

ali80, 2021-03-29:
setting `lib_ldf_mode=deep` may slow things down, though I'm not sure.
cloning `ESP Async WebServer` repo in `lib` folder and not using `platformio.ini` for this dependency fixed it for me

nicola-lunghi, 2021-11-24:
I think this library needs the following libraries to be specified as dependencies
FS
Wifi

alisherWork, 2023-11-17:
Special thanks to the seniors. This problem has been giving me a headache for a week. I solved it with one line using lib_ldf_mode=deep. My name is ali. One day I will help you too.

0blu, 2024-03-12:
Just in case anyone needs a complete `platform.ini` with `lib_ldf_mode=deep`
```ini
[env:esp32doit-devkit-v1]
platform = espressif32@3.5.0
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
lib_ldf_mode=deep
lib_deps =
    me-no-dev/AsyncTCP @ 1.1.1
    me-no-dev/ESP Async WebServer @ 1.2.3
    arduino-libraries/Arduino_JSON @ 0.2.0
```