me-no-dev/ESPAsyncWebServer #521: Catastrophic heap loss (17k in 127 mS) with consecutive requests

philbowles, 2019-05-30:
# H/W: 
ESP8266 (various: same behaviour on Wemos D1. nodeMCU, SONOFF basic etc)
# S/W: 
ESPAsyncWebServer-master at version 1.2.0
ESPAsyncTCP-master at version 1.2.0

Arduino IDE 1.8.9 / ESP8266 core 2.5.2 (same problem on 2.5.0 and 2.4.2)

# Problem
Rapid consecutive requests cause massive - and fatal - heap depletion. The code in my project is 4500+ lines in 20 source files, so all I can do is give a scenario and show the debug traces:

# Explanation of diagnostics
Scenario: App delivers webpage containing js that opens an SSE object to receive async push broadcasts. Webpage JS also makes both timer ajax calls (e.g. periodic "keepalive ping" about every 30 seconds) and user generated requests.

The APP makes continual multiple rapid SSE push broadcasts (these are "throttled" to prevent heap depletion in the SSE queue) (See: also https://github.com/me-no-dev/ESPAsyncWebServer/pull/509 which sadly has had no response as yet)

I have added a diagnostic line to Webserver.cpp to show the heap when onClient occurs:

```cpp
  _server.onClient([](void *s, AsyncClient* c){
    if(c == NULL)
      return;
    Serial.printf("T=%d FH=%d _server.onClient %08x %s\n",millis(),ESP.getFreeHeap(),c,c->remoteIP().toString().c_str());
    c->setRxTimeout(3);
    AsyncWebServerRequest *r = new AsyncWebServerRequest((AsyncWebServer*)s, c);
    if(r == NULL){
      c->close(true);
      c->free();
      delete c;
    }
  }, this);
```
# The logs
Initially we have the heap hovering around 20 - 21k and as each ping comes in, it dips to about 16-17k as the reply is generated. In terms of data the incoming /ajax/ping has none and the reply is a tiny JSON object {"ajax": "OK"}
```
T=18734567 FH=20720 _server.onClient 3fff6a2c 192.168.1.20
T=18734573 FH=17848 HTTP: /ajax/ping/ from 192.168.1.20
T=18764563 FH=18976 _server.onClient 3fff695c 192.168.1.20
T=18764568 FH=16104 HTTP: /ajax/ping/ from 192.168.1.20
```
Also you will see the SSE retry requests , e.g:
```
T=18867642 FH=21336 _server.onClient 3fff7404 192.168.1.20
T=18867763 FH=20272 _evts->onConnect 3fff756c LID=18866389 was alive 14614 cma=14
T=18881443 FH=17472 _server.onClient 3fff6f84 192.168.1.20
T=18881564 FH=20032 _evts->onConnect 3fff643c LID=18879549 was alive 13160 cma=13
```
LID is the lastID header value (in millis() ticks) I save this and calculate the difference from the previous to monitor how frequently the retries occur.  to get rough seconds which is added to a continuous moving average . cma is the secs value to add, which is delta mS/1000

We are in this "stable" condition for anything up to 8, 10 hours; sometimes as little as 30 minutes, until an accident of timing causes two requests in very close proximity, and then a crash is inevitable. The current log has been running for about 5 hours, so I gave it some rapid user activity to precipitate the problem and provide useful evidence:
```
T=19114000 FH=18696 _server.onClient 3fff6d74 192.168.1.20
T=19114006 FH=15760 HTTP: /ajax/dpin/ from 192.168.1.20
T=19121801 FH=18584 _server.onClient 3fff7064 192.168.1.20
T=19121806 FH=15712 HTTP: /ajax/ping/ from 192.168.1.20
T=19126965 FH=18792 _server.onClient 3fff6d8c 192.168.1.20
T=19126971 FH=15856 HTTP: /ajax/dpin/ from 192.168.1.20
T=19130440 FH=18560 _server.onClient 3fff70d4 192.168.1.20
T=19130443 FH=18088 _server.onClient 3fff71b4 192.168.1.20 <--- imminent disaster
T=19130448 FH=15152 HTTP: /ajax/dpin/ from 192.168.1.20
T=19130455 FH=12864 HTTP: /ajax/dpin/ from 192.168.1.20
T=19130562 FH=4152 _server.onClient 3fff9154 192.168.1.20
T=19130568 FH=1168 HTTP: /ajax/dpin/ from 192.168.1.20
```

The /ajax/dpin request is in response to a "mouseenter" event in the js. It sends a gpio number and the code responds with a block of "tooltip" text showing Digital / physical number and descriptive text, like so, i.e minimal data is transferred

[image]

As can be seen from the timings a "normal" response is dispatched within 5 or 6mS and the heap usage pattern is: about 18k at connect time, dropping to 15.7k ish immediately prior to the response. It appears as if my code takes 2k-3k heap per request / response pair, successfully and repeatedly at high speed for 5+ hours, until...

@ T=19130443 we have a second incoming connect only 3ms after the previous, with no intervening response yet...leading to 

# Question 1:
## Is this expected async behaviour, or is this "overlapping" the result of a synchronisation failure in asyncTCP and/or asyncWebserver?

My 2x consequent "user code" responses occur within 7mS of each other with an apparent additional (unsurprising) extra 2-3k heap "penalty" for the second..,. Next though we have the problem:

Within 107mS of the 2nd response, the library code has "lost" over 8k of heap. My own code then uses its 2-3k bringing us down to 1k free heap, and very shortly afterwards...of course ...crash. 

# Question 2:
## What causes the 8k loss? Is this an inevitable consequence of very closely-timed requests, or a bug that can be fixed? 

I see that other people also have this kind of issue, with no action as yet and that concerns me (https://github.com/me-no-dev/ESPAsyncWebServer/issues/200) . In fact the only response to that was to implement a fix from 2017 that was raised by...me (which would surely have already been in the failing code)

# Question 3:
## How easy / possible is it to allow the setting of a threshhold below which request will be "gracefully" rejected (e.g. HTTP 429 + retry-after header) ?

I also thought of a dynamic user boolean "go / no go" callback from onClient so that the user can choose his/ her own criteria for not going forward with the request...

# Question 4: How do the well-known projects that use the lib cope with this issue? 

Or do they never spot it because they are always lightly loaded?

In any case, I cannot use the library in "production" unless there is 

- a fix 
- a workaround 

I also cannot see that its anything I'm doing different / special: if I'm right this a fairly fundamental flaw where upon an occasional but certainly not "rare" timing condition during "heavy loading" the library will crash the app with no warning or possibility of prior prevention...

philbowles, 2019-06-14:
Greatly disappointed to see not even a comment on this, let alone a reply. I have gathered more information confirming this behaviour. Two requests in rapid succession cause memory leak of about 2k. Should I post the evidence here?

barbiani, 2019-06-14:
The project seems to have died. You will find many people reporting problems with the async library. It is fast but not very stable.

philbowles, 2019-06-14:
@barbiani I'm really sad to hear that - but thankful for your reply, I was beginning to think there was no-one there at all!

philbowles, 2019-06-14:
Looks like I will just have to fix it myself. I already fixed one other problem with it!

barbiani, 2019-06-14:
@philbowles Ever tried cancelling a transfer in  progress?

me-no-dev, 2019-06-22:
AsyncTCP has been updated. Please give it a try.

me-no-dev, 2019-06-22:
hi. disappointed or not, this is personal project of mine that I take care of only in my personal free time, which for more than a year has been none. Looking at the age of your issues, you are actually lucky. This is some code that I have shared for all to use. It is used in many projects and it has had it's flaws.
Have to say that 8266 was not one of them. The folks that maintain it have updated LwIP not long ago and that might have caused issues and to require changes. Now I can not promise when I'll have time to hook up 8266 and go through the whole install and refresh process. The issue in any case is not here. It's either in AsyncTCP or in LwIP. The effects that you see are not expected. You did not say much on what the handler actually does (/ajax/dpin/)

closed 2019-06-27 by philbowles