me-no-dev/ESPAsyncWebServer #523: AsyncTCP.h: No such file or directory, call of overloaded 'setup(int&)' is ambiguous

aspiringguru, 2019-06-12:
I'm using ESP32 Firebeetle on Windows 10 (build 17134.765) with arduino 1.8.9 (windows store 1.8.21.0)

having trouble getting ESPAsyncWebServer to work. appears to need AsyncTCP-master installed as well, but still giving me problems. 

pity as I liked the example code I found, tidier than alternative methods running webserver on the ESP32.

#include "ESPAsyncWebServer.h"

attempted to install using arduino library manager ie: arduino > menu > sketch > include library > manage libraries > search for ESPAsyncWebServer

this found PathVariableHandlers which claims to be compatible with ESPAsyncWebServer for ESP32 & ESP8266WebServer for ESP8266. 
https://github.com/sidoh/path_variable_handlers

above does not solve the "ESPAsyncWebServer.h: No such file or directory"

a search for ESPAsyncWebServer.h found 
https://github.com/me-no-dev/ESPAsyncWebServer
downloaded the zip file, unzipped to blah/Aarduino/libraries
compile produced 

ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory

found this repo > https://github.com/me-no-dev/AsyncTCP
downloaded the zip file and unzipped to blah/Aarduino/libraries

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "FireBeetle-ESP32, 80MHz, 115200"

D:\2018_working\coding\arduinoJune2019\dht22_chart\dht22_chart.ino: In function 'void setup()':

dht22_chart:25:19: error: call of overloaded 'setup(int&)' is ambiguous

   dht.setup(dhtPin);
                   ^
In file included from blah\arduinoJune2019\dht22_chart\dht22_chart.ino:12:0:

blah\Arduino\libraries\DHT_sensor_library_for_ESPx/DHTesp.h:125:8: note: candidate: void DHTesp::setup(uint8_t)

   void setup(uint8_t dhtPin) __attribute__((deprecated));

        ^

blah\Arduino\libraries\DHT_sensor_library_for_ESPx/DHTesp.h:126:8: note: candidate: void DHTesp::setup(uint8_t, DHTesp::DHT_MODEL_t)
   void setup(uint8_t pin, DHT_MODEL_t model=AUTO_DETECT);

philbowles, 2019-06-15:
"appears to need AsyncTCP-master" === DOES need AsyncTCP. 

Excuse the dumb question, but... At what point did you install AsyncTCP ? ArduinoIDE does not "pull in" dependencies, you have to install them yourself

aspiringguru, 2019-06-15:
omg. Now I can't even imagine why it didn't work. I had been downloading the git repo into arduino libraries, unzipping the .zip file &/or using the Arduino menu > sketch > library manager. I suspect using the arduino library manager to install some packages might be related but that's a guess.

I ended up uninstalling Arduino including removing the arduino folders left behind for a clean install.

It works now, still not sure exactly what I was doing to cause the problem. Probably a massive brain fade in there somewhere.

closed 2019-06-15 by aspiringguru