me-no-dev/ESPAsyncWebServer #524: Is there any way to check, using code, if the server hanged and is not responding ?

GeorgeFlorian, 2019-06-12:
Hello !

As the title says I am curious to know if there's any way of checking the status of the Web Server and in case it hanged, bugged or it's not responding I can insert:
```
server.reset();
delay(100);
ESP.restart();
```

me-no-dev, 2019-06-23:
which ESP?

GeorgeFlorian, 2019-06-23:
> which ESP?

ESP32 DevKit V4 Wrover-B

me-no-dev, 2019-06-23:
so... unless a custom WDT is implemented (which is a waste of resources) I have added (today) WDT support to AsyncTCP. You just need to enable WDT on core1 in setup. https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L89
And it will reboot the ESP is the thread is locked for more than 5 seconds (default WDT timeout)

GeorgeFlorian, 2019-06-23:
> so... unless a custom WDT is implemented (which is a waste of resources) I have added (today) WDT support to AsyncTCP. You just need to enable WDT on core1 in setup. [...]

It's the first time I'm seeing that method.
Basically, what I need to do is to write `enableCore1WDT();` anywhere in `void setup()` , preferably at the beginning of it, and that's all, right ?

me-no-dev, 2019-06-23:
yes :) that is it (and update to latest AsyncTCP of course)

GeorgeFlorian, 2019-06-24:
> yes :) that is it (and update to latest AsyncTCP of course)

Soo, adding `enableCore1WDT();` to my code makes the ESP32 restart every few seconds.

```
0x400eb1eb: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 930
0x400817e9: spi_flash_mmap_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/include/soc/soc_memory_layout.h line 173
0x400e9842: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1736
0x4008a6fd: multi_heap_free_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 375
0x400887fd: vTaskPrioritySet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/include/freertos/portable.h line 208
```

```
E (18005) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18005) task_wdt:  - IDLE1 (CPU 1)
E (18005) task_wdt: Tasks currently running:
E (18005) task_wdt: CPU 0: IDLE0
E (18005) task_wdt: CPU 1: loopTask
E (18005) task_wdt: Aborting.
abort() was called at PC 0x400eb1eb on core 0
```

My code is unusable if I add `enableCore1WDT();`.

me-no-dev, 2019-06-24:
so WDT is disabled in order to "help" run some code that is never meant to run on RTOS chip. If WDT is triggered by the loop() then you are doing something wrong there and you are holding the second CPU. See what you are doing that causes it and fix it :)

GeorgeFlorian, 2019-06-24:
> so WDT is disabled in order to "help" run some code that is never meant to run on RTOS chip. If WDT is triggered by the loop() then you are doing something wrong there and you are holding the second CPU. See what you are doing that causes it and fix it :)

The `loop` does nothing on the first run. All the code that's inside the `void loop()` is also inside an `if` condition. The `if` checks if the ESP32 is in AP_MODE and if it is in AP_MODE it will skip the entire code.
In my project, on first boot, the ESP32 is supposed to be in AP_MODE so that it can be configured. But I can't configure it because of the WTD. And I can't fix the hangs of the ATCP without a WDT.

I am out of ideas. This is uncharted territory for me.

me-no-dev, 2019-06-24:
maybe post some code? there is definitely something holding the thread

GeorgeFlorian, 2019-06-24:
Ok ! Here is the loop.
So, on first boot, and in this specific case, the following flags are set like this:
 * `restartFlag` is false
 * `AP_flag` is false

```
unsigned int free_spaces = 0;
String arrow = "";
String URL = "INVALID_URL";
String Brightness = "5";
int oldBright = 0;
String urlRefreshRate = "3";
bool AP_flag = false;
bool restartFlag = false;

int oldSignPlace = 0;
int httpFailCounter = 0;
int urlFailCounter = 0;

void loop() {
  if (restartFlag) {
    restart_sequence(5);
  }
  if(!AP_flag) { // Check if we are in AP_MODE
    int signPlace = 0;
    
    if (WiFi.status() == WL_CONNECTED && SPIFFS.exists("/configFile.txt")) {  //Check the current connection status
      logOutput((String)"IP ADDRESS: " + IP_copy.toString());
      if(URL.length() > 15) {
        HTTPClient http;
        Serial.println((String)"DEBUG: URL = " + URL);
        logOutput(http.begin(URL) ? "DNS Resolved" : "DNS Failed");
        int httpCode = http.GET(); //Make the request          
        if (httpCode > 0) { //Check for the returning code          
          logOutput((String)"[HTTP] GET... code: " + httpCode);
          if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                logOutput(payload);
                signPlace = getPlaces(payload);
                oldSignPlace = signPlace;
                httpFailCounter = 0;
                if(arrow == "") {
                  urlFailCounter ++;
                } else {
                  urlFailCounter = 0 ;
                }
          }
        } else {
            logOutput((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
            httpFailCounter++;
          }  
        http.end(); //Free the resources
        } else {
            logOutput("URL Link is invalid ! Please enter another URL");
        }
      if(urlFailCounter > 1) {
        SPIFFS.remove("/configFile.txt");
        scroll_text("URL Body format incorrect ! Couldn't find parking direction-sign !",LED_RED_HIGH,20);
        delay(100);
        restart_sequence(5);
      }
      if(httpFailCounter > 4) {
        delay(100);
        ESP.restart();
      }
      if(Brightness != NULL && Brightness.length() > 0 && Brightness.toInt() != oldBright) {
        oldBright = Brightness.toInt();
        matrixLayer.setBrightness((oldBright*255)/100);
      } 
      if(urlRefreshRate.toInt()*1000 >= 1000) {
        delay(urlRefreshRate.toInt()*1000);
      } else {
        delay(3000);
      }
    } // if (WiFi.status() == WL_CONNECTED)

    if(signPlace == 0) {
      signPlace = oldSignPlace;
    }
    if(signPlace == 2) {  // #>
          if (arrow == "<") {
            go_left(free_spaces, arrow);
          } else {
            go_right(free_spaces, arrow);
          }    
    } else if (signPlace == 3) { // <#
            if(arrow == ">") {
              go_right(free_spaces, arrow);
            } else {
              go_left(free_spaces, arrow);
            }
    }
  } // if(!APflag)
}
```

GeorgeFlorian, 2019-06-24:
UPDATE: 
Continuing my digging around the WTD I found that I can use `yield()` or `esp_task_wdt_feed()`.
So, in `void loop()` I added the following:
```
if(!AP_flag) {
} else {
yield();
//  esp_task_wdt_feed();
//  esp_task_wdt_reset();
}
```
`esp_task_wdt_feed()` doesn't help and it looks like it's deprecated.
`esp_task_wdt_reset();' doesn't help.
`yield()` doesn't to help improve the situation.

I've also added some `Serial.println()` inside the loop:
```
void loop() {
  Serial.println("Start of loop");
  Serial.println(AP_flag);
  if(!AP_flag) { // Check if we are in AP_MODE
/...
  } else {
    // yield();
    Serial.println("Do I get here?");
    esp_task_wdt_reset();
  } // if(!APflag)
```

And I don't seem them in Serial Monitor. It's like the code doesn't enter `void loop()`...

GeorgeFlorian, 2019-06-24:
UPDATE: 

The problem isn't in `void loop()`

The problem is in a `while`.
I'm using a function that starts the AP:
```
void startAP() {  
  delay(50);
  logOutput((String)"Starting AP ... ");
  logOutput(WiFi.softAP(ssidAP, passwordAP) ? "Ready" : "WiFi.softAP failed ! (Password must be at least 8 characters long )");
  delay(500);
  logOutput((String)"Setting AP configuration ... ");
  logOutput(WiFi.softAPConfig(local_IP_AP, local_IP_AP, subnetAP) ? "Ready" : "Failed!");
  delay(500);
  logOutput((String)"Soft-AP IP address: ");
  logOutput(WiFi.softAPIP().toString());
  // digitalWrite(LED, HIGH);
  delay(500);
  while (WiFi.softAPgetStationNum() == 0) {
    matrix->clear();
    matrix->setFont(NULL);
    matrix->setCursor(1,8);
    matrix->setTextColor(LED_PURPLE_HIGH);
    matrix->print((String)ssidAP);   
    matrix->setTextColor(LED_CYAN_HIGH);
    matrix->setCursor(1,0);
    matrix->print("Connect to");
    matrix->setCursor(1,16);
    matrix->print("Wireless");
    matrix->setCursor(1,24);
    matrix->print("Network"); 
    matrix->show();
  }
```

This is where the WDT gets caught up and reboots the ESP.

me-no-dev, 2019-06-24:
just add a dealy(2) in the loop and you will be fine.

GeorgeFlorian, 2019-06-24:
> just add a `delay(2)` in the loop and you will be fine.

You are a savior !
Wow. Is it that simple ?
I've tried adding `yield()` and `esp_task_wdt_feed();` and `esp_task_wdt_reset();` and nothing worked.
Except, it seems, a measly `delay()`.

Is there any documentation on this ? Because I just can't understand how methods/functions designed specifically for the purpose of resetting the WDT don't work, and a simple `delay` does.

me-no-dev, 2019-06-24:
delay will let other tasks on the core to run. this is your issue, you are not leeting other things to run on that core (the idle task that feeds the WDT has lowest priority) Feeding the WDT will not do that :) and calling wdt_feed from a task that is not the idle task, will not help much here

closed 2019-06-24 by me-no-dev