me-no-dev/ESPAsyncWebServer #527: MEMORY LEAK: 2152 bytes lost every time two requests occur in rapid succession

philbowles, 2019-06-14:
I have already raised this: disappointed to see not even a comment on the first issue report:  https://github.com/me-no-dev/ESPAsyncWebServer/issues/521

Anyway. My code is sitting in a loop doing nothing except logging to MQTT (and then into MySQL db) the value of the heap 1x per second, and logging also any incoming request.

Two types of request occur 1) an AJAX keepalive "ping" from the first (and only) page to have been served. It simply replies with a tiny message saying: {"ajok": {"ping", "ok"}}

A typical example:
```
89,"2019-06-14 18:01:45",16704,NULL
90,"2019-06-14 18:01:46",16704,NULL
91,"2019-06-14 18:01:48",16704,NULL
92,"2019-06-14 18:01:48",15416,"NEW TASK=3fff7d84 TSS=3fff7cfc rmin=0 rmax=0 /ajax/ping/"
93,"2019-06-14 18:01:48",14240,"END TASK=3fff7d84 TSS=3fff7cfc rmin=0 rmax=0 nrq=0 /ajax/ping/"
94,"2019-06-14 18:01:48",14240,"DIE TASK=3fff7d84 TSS=3fff7cfc rmin=0 rmax=0 nrq=0 /ajax/ping/"
95,"2019-06-14 18:01:48",16704,NULL
96,"2019-06-14 18:01:50",16704,NULL
97,"2019-06-14 18:01:51",16704,NULL
```

As can be seen the heap before and after is unchanged. This is the same pattern for every occurrence about every 30 seconds. There is only one single such event in the whole log where the heap is lower after the call, and it occurs immediate;y before the 2k lea - see later.

The other type of request is a "upnp" status request from an amazon echo dot, and occurs about every 10 minutes. There is no on("upnp"... handler. Theses message are managed by the onNotFound handler, which simply replies with  		request->send(404, "text/plain","NO!"); In the log, they like this:
```
512,"2019-06-14 18:08:21",16552,NULL
513,"2019-06-14 18:08:22",16240,NULL
514,"2019-06-14 18:08:22",12976,"NEW TASK=3fff8904 TSS=00000000 rmin=0 rmax=0 /upnp/control/basicevent1"
515,"2019-06-14 18:08:22",13928,"END TASK=3fff8904 TSS=00000000 rmin=0 rmax=0 nrq=0 /upnp/control/basicevent1"
516,"2019-06-14 18:08:23",14400,NULL
517,"2019-06-14 18:08:24",14400,NULL
518,"2019-06-14 18:08:25",14400,NULL

```
In these circumstances the heap does not immediately recover, It can take several minutes to regain its former level, and is usually a few bytes short
```
686,"2019-06-14 18:11:03",14344,NULL
687,"2019-06-14 18:11:04",14344,NULL
688,"2019-06-14 18:11:04",10896,"NEW TASK=3fff9504 TSS=3fff947c rmin=0 rmax=0 /ajax/ping/"
689,"2019-06-14 18:11:04",10392,"END TASK=3fff9504 TSS=3fff947c rmin=0 rmax=0 nrq=0 /ajax/ping/"
690,"2019-06-14 18:11:04",10392,"DIE TASK=3fff9504 TSS=3fff947c rmin=0 rmax=0 nrq=0 /ajax/ping/"
691,"2019-06-14 18:11:05",16496,NULL
692,"2019-06-14 18:11:06",16496,NULL
```

We seem to have lost 56 bytes (16552 - 16496)

This continues for a couple of hours: with each 404 handler call losing  between 32 and 56 bytes until the BIG problem where 2152 bytes disappears never to return. It appears in the red circle where we get the "double-dip" and after that "recovery" to about 14k the previous level never returns. When left for many hours - doing nothing except replying with very short message to occasional requests - gradually it will run out of heap then crash.

The problem starts at 20:11:17.
```
8417,"2019-06-14 20:11:17",16424,NULL
8418,"2019-06-14 20:11:17",14320,"NEW TASK=3fff7dac TSS=3fff7d44 rmin=0 rmax=0 /ajax/ping/"
8419,"2019-06-14 20:11:17",13960,"END TASK=3fff7dac TSS=3fff7d44 rmin=0 rmax=0 nrq=0 /ajax/ping/"
8420,"2019-06-14 20:11:17",13960,"DIE TASK=3fff7dac TSS=3fff7d44 rmin=0 rmax=0 nrq=0 /ajax/ping/"
8421,"2019-06-14 20:11:18",16112,NULL
8422,"2019-06-14 20:11:18",12848,"NEW TASK=3fff7b04 TSS=00000000 rmin=0 rmax=0 /upnp/control/basicevent1"
8423,"2019-06-14 20:11:18",13800,"END TASK=3fff7b04 TSS=00000000 rmin=0 rmax=0 nrq=0 /upnp/control/basicevent1"
8424,"2019-06-14 20:11:19",14272,NULL
8425,"2019-06-14 20:11:20",14272,NULL
8426,"2019-06-14 20:11:21",14272,NULL
8427,"2019-06-14 20:11:22",14272,NULL
8428,"2019-06-14 20:11:23",14272,NULL
8429,"2019-06-14 20:11:24",14272,NULL
8430,"2019-06-14 20:11:25",14272,NULL
```
The ping does not recover to the exact pre-call value. This is the ONLY point in the log where this occurs and - not coincidentally - it is followed almost immediately by a 404 handler, after which the heap never recovers back past 14272. 

[image]

This pattern of failure is absolutely consistent: when two requests arrive within less than a second of each other, there is a memory leak of about 2k. If three come in, rapidly 4k is lost, i.e. for every additional request > 1 in a "rapid set", 2k is leaked.

philbowles, 2019-06-14:
This might help: I put some diagnostics in, and got lucky quickly:  when there is an "overlap" in AsyncWebserverRequests, i.e a 2nd one arrives before the first is complete, the destructor for the first is not called

```
T=112624 FH=17712 _server.onClient 3fff7d34 192.168.1.8
T=112625 FH=17168 AsyncWebServerRequest CTOR <----------------------------------- 1 IN 
NEW TASK=3fff8abc TSS=00000000 rmin=0 rmax=0 /upnp/control/basicevent1
T=112633 FH=15488 HTTP: /upnp/control/basicevent1 from 192.168.1.8
END TASK=3fff8abc TSS=00000000 rmin=0 rmax=0 nrq=0 /upnp/control/basicevent1
T=139674 FH=15536 _server.onClient 3fff8a24 192.168.1.20
T=139676 FH=14992 AsyncWebServerRequest CTOR <-------------------------------------- 2 IN overlaps 1
NEW TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 /ajax/ping/
T=139684 FH=12088 HTTP: /ajax/ping/ from 192.168.1.20
END TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
DIE TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
T=139798 FH=14464 AsyncWebServerRequest DTOR <---------------------------------------- 2 OUT
T=170517 FH=15560 _server.onClient 3fff793c 192.168.1.20
T=170518 FH=15016 AsyncWebServerRequest CTOR <---------------------------------------- 3 IN
NEW TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 /ajax/ping/
T=170526 FH=12088 HTTP: /ajax/ping/ from 192.168.1.20
END TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
DIE TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
T=170635 FH=14464 AsyncWebServerRequest DTOR <---------------------------------------- 3 OUT
```

Prima Facie case of Memory Leak by sizeof(AsyncWebServerRequest)

philbowles, 2019-06-14:
This might help: I put some diagnostics in, and got lucky quickly: when there is an "overlap" in AsyncWebserverRequests, i.e a 2nd one arrives before the first is complete, the destructor for the first is not called immediately. Other tests show that is can occur some minutes later - what's holding on to it? But it seems there are occasions when that delayed destrcutor does not get called at all, and thats when we lose the memory.

T=112624 FH=17712 _server.onClient 3fff7d34 192.168.1.8
T=112625 FH=17168 AsyncWebServerRequest CTOR <----------------------------------- 1 IN 
NEW TASK=3fff8abc TSS=00000000 rmin=0 rmax=0 /upnp/control/basicevent1
T=112633 FH=15488 HTTP: /upnp/control/basicevent1 from 192.168.1.8
END TASK=3fff8abc TSS=00000000 rmin=0 rmax=0 nrq=0 /upnp/control/basicevent1
T=139674 FH=15536 _server.onClient 3fff8a24 192.168.1.20
T=139676 FH=14992 AsyncWebServerRequest CTOR <-------------------------------------- 2 IN overlaps 1
NEW TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 /ajax/ping/
T=139684 FH=12088 HTTP: /ajax/ping/ from 192.168.1.20
END TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
DIE TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
T=139798 FH=14464 AsyncWebServerRequest DTOR <---------------------------------------- 2 OUT
T=170517 FH=15560 _server.onClient 3fff793c 192.168.1.20
T=170518 FH=15016 AsyncWebServerRequest CTOR <---------------------------------------- 3 IN
NEW TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 /ajax/ping/
T=170526 FH=12088 HTTP: /ajax/ping/ from 192.168.1.20
END TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
DIE TASK=3fff8f4c TSS=3fff8ec4 rmin=0 rmax=0 nrq=0 /ajax/ping/
T=170635 FH=14464 AsyncWebServerRequest DTOR <---------------------------------------- 3 OUT

barbiani, 2019-06-14:
Maybe chrome requesting the index.html and favicon will break the server
after a few reloads.

philbowles, 2019-06-14:
@barbiani - no, I have every single incoming request logged to mysql. There are only the two types I mention. No others.

GeorgeFlorian, 2019-06-20:
Don't bother ! This library has no support whatsoever. It has critical flaws, it crashes without any way of recuperating.

me-no-dev, 2019-06-22:
Update has been pushed to AsyncTCP that should fix many of the memory issues (if not all).

me-no-dev, 2019-06-22:
so, the fact that you lose 30-52 bytes + some that come back after some time (2 minutes) every OnNotFound means that the response was not formatted correctly, the connection was not properly cleared, so it was put into TIME_WAIT for 2 minutes, after which it was cleared, but something left dangling (30-52 bytes)
Honestly do not know how to help. This same code is fine on ESP32 and the web server is shared. Different are the TCP layers.

philbowles, 2019-06-27:
Not sure how I better format the response:  request->send(404, "text/plain","NO!"); .

Is there any way I can exercise control over "the connection was not properly cleared" - happy to try anything!

closed 2019-10-05 by stale[bot]

keithsw1111, 2023-07-29:
I also encountered leaks but I think it was because I was calling request->send() inside the filehandler. It looks like calling this in the request handler is the only valid place to call it.