me-no-dev/ESPAsyncWebServer #528: MEMORY LEAK in onNotFound handler: repeatable MCVE

philbowles, 2019-06-15:
I'm raising another issue in the hope that someone might actually DO something, but this is the same problem as https://github.com/me-no-dev/ESPAsyncWebServer/issues/527 and https://github.com/me-no-dev/ESPAsyncWebServer/issues/521 which have received no attention. I have now narrowed it down further and provided an MCVE which shows the problem and is consistently repeatable.

```cpp
#include <ESP8266WiFi.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "LaPique";
const char* password = "CanonD60";

AsyncWebServer server(80);
void* grab;

void setup(void) {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  WiFi.waitForConnectResult();

  server.onNotFound([](AsyncWebServerRequest* request){   
    request->send(404, "text/plain","NO!");
  });
  
  server.begin();
  
  // grab 80% of the heap to speed things up
  grab=malloc((ESP.getFreeHeap() * 80) / 100);
  Serial.printf("LOW HEAP = %d\n",ESP.getFreeHeap());
}

void loop(void) {}
```

I then make repeated browser requests from 192.168.1.20 while causing my amazon echo dot on 192.168.1.8 to make simultaneous requests. On a low heap, it doesn't take long to show the problem. I have added 3 diagnostic lines to the library, whose content should be obvious:

```
LOW HEAP = 9664
ESPAsyncWebServer server started
T=15743 FH=8200 _server.onClient 3fffa094 192.168.1.20
T=15744 FH=7656 AsyncWebServerRequest 3fffa174 CTOR sizeof=284
T=15745 FH=7528 _server.onClient 3fffa294 192.168.1.20
T=15748 FH=6984 AsyncWebServerRequest 3fffa374 CTOR sizeof=284
T=15801 FH=6488 AsyncWebServerRequest DTOR 3fffa374
T=16037 FH=7520 _server.onClient 3fffa294 192.168.1.20
T=16038 FH=6976 AsyncWebServerRequest 3fffa374 CTOR sizeof=284
T=16159 FH=6488 AsyncWebServerRequest DTOR 3fffa174
T=16703 FH=7520 _server.onClient 3fffa01c 192.168.1.20
T=16704 FH=6976 AsyncWebServerRequest 3fffa0fc CTOR sizeof=284
T=16825 FH=6488 AsyncWebServerRequest DTOR 3fffa374
T=17183 FH=7520 _server.onClient 3fffa294 192.168.1.20
T=17184 FH=6976 AsyncWebServerRequest 3fffa374 CTOR sizeof=284
T=17249 FH=3896 _server.onClient 3fffb084 192.168.1.8
T=17250 FH=3352 AsyncWebServerRequest 3fffb164 CTOR sizeof=284
T=17303 FH=4336 AsyncWebServerRequest DTOR 3fffa0fc
T=17668 FH=2616 _server.onClient 3fffa01c 192.168.1.20
T=17669 FH=2072 AsyncWebServerRequest 3fffa6ec CTOR sizeof=284
T=17721 FH=4336 AsyncWebServerRequest DTOR 3fffa374
T=20993 FH=5872 AsyncWebServerRequest DTOR 3fffa6ec
T=27263 FH=5872 _server.onClient 3fffade4 192.168.1.20
T=27264 FH=5328 AsyncWebServerRequest 3fffaec4 CTOR sizeof=284
T=27266 FH=5152 _server.onClient 3fffa664 192.168.1.20
T=27268 FH=4608 AsyncWebServerRequest 3fffa744 CTOR sizeof=284
T=27321 FH=4168 AsyncWebServerRequest DTOR 3fffaec4
T=27762 FH=5200 _server.onClient 3fffade4 192.168.1.8
T=27763 FH=4656 AsyncWebServerRequest 3fffaec4 CTOR sizeof=284
T=28040 FH=408 _server.onClient 3fff9ff4 192.168.1.20

Exception (29):
epc1=0x40204195 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fffec10 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffedb0:  38323d54 20303430 343d4846 5f203830  
3fffedc0:  76726573 6f2e7265 696c436e 20746e65  
3fffedd0:  66666633 34666639 32393120 3836312e  
3fffede0:  322e312e 00000a30 3fffee50 40206a7b  
3fffedf0:  3fffee30 3fffee00 00000008 40206aad  
3fffee00:  3ffef37c 0000179d 0000179d 4010020c  
3fffee10:  3ffef16c 00000198 3fffee50 00006d88  
3fffee20:  3ffef16c 00000198 0000011c 401004e4  
3fffee30:  2e323931 2e383631 0000011c 00006d88  
3fffee40:  3ffef16c 00000000 3fff9ff4 402054d5  
3fffee50:  00000000 00000000 00000000 40207ae8  
3fffee60:  1401a8c0 00000030 0000000b 3ffef61c  
3fffee70:  3fffaa0c 3ffef16c 000000d8 3ffef61c  
3fffee80:  3fff9ff4 3ffef16c 3fffaa0c 40205e8a  
3fffee90:  00000000 00000000 3fffaa0c 402060a4  
3fffeea0:  3fffaa0c 3ffef600 3ffef601 402060d0 
... etc
```

Interestingly, if I remove the server.onNotFound , I can't make it happen. I also cannot make it happen, no matter how fast I hit "refresh" if I only send in requests from a single IP.

With the explicit handler code included and using two IPs, I  can make it happen within seconds, every time...(within many minutes unless I shrink the heap first, but the same behaviour exists an all sizes of heap.)

The preconditions for the memory leak condition seems to be:
1) An explicit handler
2) simultaneous requests from DIFFERENT IPs

I have spent literally WEEKS trying to isolate this problem. I know the repo owner has other things in his life than this, but PLEASE PLEASE PLEASE can someone at least TRY to look at this and fix it?

GeorgeFlorian, 2019-06-20:
Don't bother !
@me-no-dev Said he knows about this but that he either has no idea how to fix it or that he doesn't find the time.

And he isn't been finding the time for the past year or so.

https://github.com/me-no-dev/ESPAsyncWebServer/issues/529

me-no-dev, 2019-06-20:
I have been finding time... no result yet. If anything it seems that 8266 has followed ESP32's issues. They both use the same LwIP... could be a clue. TBH any time I need a server, i use Async, just to give it more chance for resolving.

Aircoookie, 2019-06-20:
@me-no-dev thank you for looking into it! From your comment it seems like you believe this issue is caused by upstream (either AsyncTCP, Arduino core or LWIP) and not by the library itself, right?

On an unrelated note, please merge https://github.com/me-no-dev/ESPAsyncWebServer/pull/491 :)

@philbowles What Arduino core and LwIP versions have you been using? If the results vary with the core used, that might give us more clues. This issue also seems to affects my projects so I'd be keen on seeing a fix!

me-no-dev, 2019-06-20:
I think that the issue generally lies in some change in LwIP between the versions. Older versions behaved differently. While I was thinking that maybe RTOS is the issue on ESP32 (I mean my code is the issue but RTOS is the driving factor), the fact that same reports started appearing on 8266 kinda points towards that...

The issue is not in the AsyncWebServer lib at all. It's in AsyncTCP :)

baggior, 2019-06-20:
> @me-no-dev 

> On an unrelated note, please merge #491 :)

me-no-dev, 2019-06-22:
merged and also AsyncTCP has been updated. Please give a try :)

Aircoookie, 2019-06-22:
@me-no-dev Awesome, thanks for the (hopefully) fix! I've never really had issues with ESP32, might be due to the larger amount of memory not showing such leaks as early as with ESP8266.

Can this fix also be applied to ESPAsyncTCP (8266 version) or do you suspect the issue on that platform has a different cause? I would be happy to help, but the TCP library code is beyond my comprehension :)

me-no-dev, 2019-06-22:
the ESP32 issue was in multithreading. ESP8266 is not multithreaded, so that will need further investigation. TBH I have not booted 8266 in a loong time... :) will try soon.

me-no-dev, 2019-06-22:
@philbowles can you also make it happen if you do not starve the memory as much? Like give it another 4-5K? Adding the callback and using different hosts will cause memory usage on their own.

Aircoookie, 2019-06-23:
@me-no-dev In my application there is a `/freeheap` page (initial memory available after boot 27kB). If I poll that once per second while having an AsyncMqttClient connection to `broker.hivemq.com` established (not even publishing anything or getting messages send to the subscribed topic), the memory decreases by around 3kB about every 15 seconds (in one go, the 15 seconds in between the number stays the same). Until, at around 3-6kB left, it wdt resets.

I believe it might be due to some connection not being closed in time or a more specific issue caused by the MQTT library (in my case) or the Echo (in philbowles case). I tried the above MCVE (just commented out the memory starvation) and I am unable to get it to leak the heap with 2 hosts sending 10 req/sec each (tried both lwip 1 and 2). It went down by 20kB at one time but then recovered to the initial 45kB ten seconds later. I can exclude weak wifi signal as a possible cause, the router is 1m away and no interfering networks around.

me-no-dev, 2019-06-23:
So since yesterday I have been testing camera streaming over Async. 25FPS+ for over 24hours. Not a byte lost. I randomly connected from another browser to loath things extra. No glitch at all :)

Aircoookie, 2019-06-23:
Alright, I spoke to soon about the MCVE. It did crash, but only after 30 min. Before that, no leak occured, but then the heap rapidly declined over 10 sec and the system crashed. It happened again and again directly after the crash (due to built-up unhandled requests during the reset i suppose)
Good to hear that the streaming worked though, I will try and see if there anything else could be causing the issue.

<details><summary>Here is the modified MCVE</summary><p>

```cpp
//Arduino 1.8.9, tested with ESP8266 core v2.5.2 Lwip 1 and 2

#include <ESP8266WiFi.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "x";
const char* password = "x";

AsyncWebServer server(80);

void setup(void) {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  WiFi.waitForConnectResult();
  Serial.println(WiFi.localIP());

  server.onNotFound([](AsyncWebServerRequest* request){   
    request->send(404, "text/plain","NO!");
    Serial.printf("LOW HEAP = %d\n",ESP.getFreeHeap());
  });
  
  server.begin();
}

void loop(void) {}
```

</p></details>

<details><summary>And the serial out</summary><p>

```
//was happily sitting at around 43kB for half an hour then:
LOW HEAP = 43720
LOW HEAP = 45304
LOW HEAP = 43720
LOW HEAP = 45584
LOW HEAP = 43720
LOW HEAP = 45480
LOW HEAP = 43720
LOW HEAP = 45320
LOW HEAP = 42776
LOW HEAP = 45320
LOW HEAP = 44320
LOW HEAP = 45320
LOW HEAP = 46248
LOW HEAP = 45920
LOW HEAP = 43704
LOW HEAP = 43720
LOW HEAP = 45584
LOW HEAP = 43720
LOW HEAP = 45464
LOW HEAP = 43720
LOW HEAP = 45320
LOW HEAP = 43720
LOW HEAP = 45320
LOW HEAP = 44320
LOW HEAP = 45320
LOW HEAP = 44320
LOW HEAP = 45320
LOW HEAP = 46248
LOW HEAP = 45272
LOW HEAP = 43040
LOW HEAP = 42120
[... 815 lines cut ...]
LOW HEAP = 40808
LOW HEAP = 39784
LOW HEAP = 39136
LOW HEAP = 37752
LOW HEAP = 37736
LOW HEAP = 38120
LOW HEAP = 38120
LOW HEAP = 37120
LOW HEAP = 36504
LOW HEAP = 36776
LOW HEAP = 34824
LOW HEAP = 33560
LOW HEAP = 34376
LOW HEAP = 35320
LOW HEAP = 32216
LOW HEAP = 33792
LOW HEAP = 34088
LOW HEAP = 34088
LOW HEAP = 33416
LOW HEAP = 32416
LOW HEAP = 30200
LOW HEAP = 31960
```

</p></details>

philbowles, 2019-06-27:
@aircookie @me-no-dev Apologies for delay, and thanks for responses. I have tried all versions of lwip above 1.4 and also core 2.4.2 and 2.5.0 - same results.

"Alright, I spoke to soon about the MCVE. It did crash, but only after 30 min. Before that, no leak occured, but then the heap rapidly declined over 10 sec and the system crashed." I'm glad it's not just me, that's exactly what happens, "out of nowhere"

@me-no-dev I depleted the heap to speed up the error log. It will do it from any heap size, its just takes longer because the "overlap"that causes it is rare and loses "only" 2k, so if I start with 28k and the "overlap" occurs once every 1/2 hour, it takes 7hours to fail.

closed 2019-10-05 by stale[bot]

cgmckeever, 2020-08-06:
> So since yesterday I have been testing camera streaming over Async. 25FPS+ for over 24hours. Not a byte lost. I randomly connected from another browser to loath things extra. No glitch at all :)

@me-no-dev do you have example code of this? I appear to be having a memory panic when I do `webSocket.binaryAll(buf, len);`