me-no-dev/ESPAsyncWebServer #529: This library is unreliable, not maintained, not supported and it doesn't work in a real life scenario

GeorgeFlorian, 2019-06-20:
Hello !

I've been trying to make this library work for the past 3 months and I couldn't !

It just does not work well ! It hangs and it crashes whenever and wherever.

It sometimes works and sometimes doesn't ! And there is no way of knowing when it doesn't !

It doesn't matter if you are in AP or STA. It just crashes on you. And when it does you have to physically press on RESET button otherwise you can't do shit with your ESP.

@me-no-dev I am not trying to attack you or be ungrateful because that would be really idiotic of me. I am not able to make a library like this one, nor I really want to learn how to. But this product of yours isn't working and you choosing to ignore is what makes me mad.

It's not that hard to add some lines into the README stating that the library is buggy and that it has critical flaws that can and will impend the working flow of the device and that you don't want to work on this project anymore on personal reasons. I understand that you don't have time or you that you don't want to, but this library has been downloaded thousands of times by users who need a reliable WebServer without knowing that this does not work.

I am not asking you to fix it, because others have and you weren't too bothered by that. I am asking you to put something in the README so that idiots like me won't have to waste time and resource trying to make this library to work in their projects.

gibo77, 2019-06-20:
It works great for me though. I used Fetch() in react native as client and ESP8266 as simple server. Look at your power supply. Make sure they are stable.

Aircoookie, 2019-06-20:
@GeorgeFlorian it's great that you show concern for this library, we all do.
Keep in mind though that all contributors, mostly me-no-dev, put a lot of work into this. It's not like you are paying for it, so if there is a better library available, then please go ahead and use that. (there isn't)
I'm also experiencing crashes, but if this library works it is amazing and fast as anything!
I'm not saying this is well maintained at the moment, but please don't spam other issues telling people to not bother. It is not constructive for anyone. If you want to help, please try and exactly determine the circumstances under which the library doesn't work for you, like the OP of #528 did.

GeorgeFlorian, 2019-06-20:
> @GeorgeFlorian it's great that you show concern for this library, we all do.
> Keep in mind though that all contributors, mostly me-no-dev, put a lot of work into this. It's not like you are paying [...]

I made this from the perspective of an idiot for other idiots. I lack the skill to help ! Even if I wanted to try I don't understand how Networking works.
On the other hand, I would pay for this library if it was reliable.
And I haven't found any other library that is recommended like this one.

closed 2019-06-21 by GeorgeFlorian