me-no-dev/ESPAsyncWebServer #532: AsyncWebserver+CaptivePortal+load index.html from SPIFFS on ESP32

williamesp2015, 2019-06-23:
I am wondering why I can not have a mixed AsyncWebserver + Captive Portal to get ID and Name send to the index.html saved on SPIFFS and then  open index.html. I am using ESP32 -Arduino platform. Any help would be greatly appreciated

me-no-dev, 2019-06-24:
could you please explain in more detail what you want to achieve?

williamesp2015, 2019-06-24:
Thanks @me-no-dev. I am desperately trying to connect APSTA Mode and connect to static IP using AP and then connect to Internet by STA. After adding Captive portal  I am not able to connect websocket and Captive portal don't show. After connect to AP, If I don't have connection to Internet, My chrome browser on WIN 10  opens "http://www.msftconnecttest.com/redirect" but not going to open Captive portal page My source is this:

#include <Arduino.h>
#include <webServer.h>
#include <ESPAsyncWebServer.h>
#include <ArduinoOTA.h>
#include <stdint.h>
#include <SPIFFS.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
AsyncWebSocket ws("/ws"); // access at ws://[esp ip]/ws
AsyncWebServer server(80);
AsyncEventSource events("/events");
IPAddress local_IP(8,8,8,8);
IPAddress gateway(5,5,5,1);//Ends with 1 with same IP initialize
IPAddress subnet(255, 255, 255, 0);
#include <DNSServer.h>
DNSServer dns_server;
#define HOSTNAME "MYwifi.net"
#define ssid "STAssid"
#define password "password12345"
class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    //request->addInterestingHeader("ANY");
    return true;
  }

  void handleRequest(AsyncWebServerRequest *request) {
    AsyncResponseStream *response = request->beginResponseStream("text/html");
    response->print("<!DOCTYPE html><html><head><title>Captive Portal</title></head><body>");
    response->print("<p>This is out captive portal front page.</p>");
    response->printf("<p>You were trying to reach: http://%s%s</p>", request->host().c_str(), request->url().c_str());
    response->printf("<p>Try opening <a href='http://%s'>this link</a> instead</p>", WiFi.softAPIP().toString().c_str());
    response->print("</body></html>");
    request->send(response);
  }
};
void STAconnect();
{
WiFi.begin(ssid, password);
while(WiFi.waitForConnectResult() != WL_CONNECTED){
       Serial.print(".");
       delay(1000);
 }
}
void onWsEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
  switch(type) {
          case WS_EVT_DISCONNECT:
             Serial.println("Client disconnected");
              break;
          case WS_EVT_CONNECT:
              {
                Serial.println("Websocket client connection received");
                client->text("Hello from Server to client:"+client->id());
                Serial.printf("ws[%s][%u] connect\n", server->url(), client->id()); 
              }
              break;
          case WS_EVT_DATA:
             Serial.println(data);
              break;
      }
  }

void setup()
{
    Serial.begin(115200);
    WiFi.mode(WIFI_MODE_APSTA);
    delay(100);
    WiFi.softAPsetHostname(HOSTNAME);
     delay(100);
     Serial.print("Setting soft-AP configuration ... ");
      Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");
      delay(100);
      Serial.print("Setting soft-AP ... ");
      WiFi.softAP("Captive Portal");
      delay(100);
      Serial.print("Soft-AP IP address = ");
      Serial.println(WiFi.softAPIP());
      delay(500);
     dns_server.start(53, "*", WiFi.softAPIP());
     server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);
     ws.onEvent(onWsEvent);
     server.addHandler(&ws);
     server.begin();
     delay(100);
     STAconnect();
}
void loop(){
  dns_server.processNextRequest();
}

me-no-dev, 2019-06-24:
8.8.8.8 for local IPs? are you expecting to be able to have internet on your laptop while you are connected to ESPs AP? Like a router?

williamesp2015, 2019-06-24:
Not on PC or laptop. I need to connect ESP32 websocket by AP and connect it to Internet by STA mode. I changed Static IP to 192.168.4.4 but same result and no connection to websocket. I uploaded my index.html on SPIFFS and previously websocket was working fine. Seems conflict with Captive portal

me-no-dev, 2019-06-24:
try in the following order:
```cpp
     ws.onEvent(onWsEvent);
     server.addHandler(&ws);
     server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);
     server.begin();
```

williamesp2015, 2019-06-24:
I appreciate your help but I don't know why can not connect to the websocket.

me-no-dev, 2019-06-25:
did you try?

williamesp2015, 2019-06-25:
I tested STA and AP mode and Websocket is nicely working but in STA or AP mode but Captive portal having problem msot of the time with below setup. Mostof the time Captive portal is not working  on PC or laptop and never worked on iphone. Some times Captive portal try to redirect but failed or couldn't connect to websocket
        WiFi.mode(WIFI_MODE_APSTA);
        delay(100);
        Serial.print("Setting soft-AP configuration ... ");
        Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");
        delay(100);
        Serial.print("Setting soft-AP ... ");
        WiFi.softAP("Captive Portal");
        delay(100);
        Serial.print("Soft-AP IP address = ");
        Serial.println(WiFi.softAPIP());
        WiFi.softAPsetHostname(HOSTNAME);
        dnsServer.start(53, "*", WiFi.softAPIP());
        ws.onEvent(onWsEvent);
        server.addHandler(&ws);
        server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=30").setDefaultFile("index.html");
        server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);
        server.begin();

boarchuz, 2019-06-26:
1. What is at 8.8.8.1? I doubt you want the gateway to be different to the AP IP. Use the same address.
2. And make it a valid IP. 8.8.8.8 is not in a private range. Especially since 8.8.8.8 happens to be a very popular Google DNS server, there's no telling how Windows/iOS will handle it.

atanisoft, 2019-06-26:
Using 8.8.8.8 is a workaround for older Android phones which hardcore this IP for connectivity checks. There is no problems in using this as the AP IP as long as internet access is not required via the portal (ie it is used as a wifi setup portal and disabled after config / reboot)

williamesp2015, 2019-06-28:
I tested Static APIP with 8.8.8.1  Websocket is working but Captive portal not always on laptop but never on iphone.

carloslema, 2019-09-24:
Thanks @me-no-dev for the CaptiveRequestHandler Class. 

I am facing similar issues as @williamesp2015 — I got stuck getting the parameters after a user submits the form with the selected network and its credentials. I am sure it is something super simple. I just can't figure it out.

The Captive Page works with a smart phone, tablet and desktop computer by connecting directly to the ESP32 or by connecting to 192.168.4.1.

Here is the working sketch — 99% is from @me-no-dev code, I just added a few things here and there.

Any help is much appreciated.  \\ carlos

```
#include <DNSServer.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ArduinoJson.h>

DNSServer dnsServer;
AsyncWebServer server(80);
AsyncWebSocket ws("/ws");

/**
 * GOAL: Get user to connect TheThing to a Wifi Access point (home, work etc)
 * 
 * User will use a smart phone, tablet or desktop computer to connect directly to TheThing for the first time, or to connect to a different network
 * TheThing will open a Captive page, where user selects a WiFi Network and enters its password.
 * After user submits the credentials, they will be rederected to a Thank You you page 
 * that contains next steps, (disconnect from the TheThing, go to the DNS name TheThing.local to start using TheThing.
 * 
 */

class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    return true;
  }

String networkName;
String password;

  String htmlHeaders    = "<meta charset=\"utf-8\"><meta http-equiv=\"X-UA-Compatible\" content=\"IE=edge\"><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, shrink-to-fit=no\"><meta http-equiv=\"Cache-Control\" content=\"no-cache, no-store, must-revalidate\" /><meta http-equiv=\"Pragma\" content=\"no-cache\" /><meta http-equiv=\"Expires\" content=\"-1\" />";

  /***
   * Hadle request
   */
  void handleRequest(AsyncWebServerRequest *request) {

    String networkName;
    String password;
    
    // Get WiFi Network Names
     int networks = WiFi.scanNetworks();

    // Stream response (build) 
    AsyncResponseStream *response = request->beginResponseStream("text/html");

    // START HTML Page
    response->print("<!DOCTYPE html><html><head>");
    
    // handle POST
    if(request->url() == "wifisave"){
        int params = request->params();

        /**
         * Keeps returning 0 params
         * Insomnia is sending 
         * { "ssid": "networkNameGoeshere", "password": "password" }
         */
[... 101 lines cut ...]
  
  WiFi.softAP("TheThing Wifi Setup");  
  dnsServer.start(53, "*", WiFi.softAPIP());

  // Get IP to check with Laptop
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

  // Order seems to matter — me-no-dev
  ws.onEvent(onWsEvent);
  server.addHandler(&ws);
     
  // Handle Captive Form
  server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER); //only when requested from AP
  Serial.println("-----------------------------| What happens after? Save to EEPROM & ReStart Deca?");

  server.begin();
}

void loop(){
  dnsServer.processNextRequest();

}
```

williamesp2015, 2019-09-26:
Thanks @carloslema. For the first time my iphone opened Captive Portal page but I SSID shows NO NETWORK FOUND. How to Enter only by click on a button and run index.html on SPIFF?

carloslema, 2019-09-26:
@williamesp2015 that is strange. The ESP32 HUZZAH that I am using only supports IEEE-802.11B, IEEE-802.11G, IEEE802.11N networks — it does not see 5G / WiFiac networks. Check what network(s) you have around. I assume you've connected the ESP32 to a WiFi network by hard coding network credentials.

About using SPIFF, it all depends on what is it you want this Captive Page to do.

You have access to: request->method() (GET, POST...etc), request->url() to route the request to the right page — like in the "if(request->url() == "/wifisave"){ // serve page here}"

Reading a file from SPIFF (SPIFFS.open(filename, "w")) shouldn't be that difficult, you can also use templates and replace placeholders (%VARNAME%) on their way out to the browser. In my case, these two pages need to be super simple (does not mean ugly, CSS can do wonders) — set up a user for the first time.

They key thing is getting a very simple example to work. Hope that helps.

williamesp2015, 2019-09-27:
I have problem to load index.html or even run http://192.168.4.1/index.html using this code:
#include <DNSServer.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ArduinoJson.h>
#include "SPIFFS.h"
#include <WiFi.h>
#include <AsyncTCP.h>
#include <DNSServer.h>
#include "ESPAsyncWebServer.h"

DNSServer dnsServer;
AsyncWebServer server(80);

class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    //request->addInterestingHeader("ANY");
    return true;
  }

  void handleRequest(AsyncWebServerRequest *request) {
    AsyncResponseStream *response = request->beginResponseStream("text/html");
    response->print("<!DOCTYPE html><html><head><title>Captive Portal</title></head><body>");
    response->print("<p>This is out captive portal front page.</p>");
    response->printf("<p>You were trying to reach: http://%s%s</p>", request->host().c_str(), request->url().c_str());
    response->printf("<p>Try opening <a href='http://%s'>this link</a> instead</p>", WiFi.softAPIP().toString().c_str());
    response->print("</body></html>");
    request->send(response);
  }
};

void setup(){
      Serial.begin(115200);
  if(!SPIFFS.begin(true)){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
  }
  // Autoconnect goes here or check EEPROM for the past 8 saved networks

  Serial.println("-----------------------------| Starting Captive Screen");
delay(300);
WiFi.softAP("MySetup");
// Get IP to check with Laptop
delay(100);
IPAddress myIP = WiFi.softAPIP();
Serial.print("AP IP address: ");
Serial.println(myIP);
dnsServer.start(53, "*", WiFi.softAPIP());
server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);//only when requested from AP
  //more handlers...
server.on("/html", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
  });
  server.on("/test.js", HTTP_GET, [](AsyncWebServerRequest *request){
     request->send(SPIFFS, "/app1.js", "text/javascript");
   });
       server.on("/test.js", HTTP_GET, [](AsyncWebServerRequest *request){
          request->send(SPIFFS, "/Setup.txt", "text/javascript");
        });
  server.begin();
}

void loop(){
  dnsServer.processNextRequest();
}

closed 2019-12-12 by stale[bot]